dresden-elektronik/deconz-rest-plugin #1072: Segmentation fault (core dumped)

martikainen87, 2018-12-28:
**Deconz GW version:** 2.05.52
**Firmware:** 262F0500 (no available update according to deconz logs)

I'm running deconz through marthoc's addon for hass.io (https://github.com/marthoc/hassio-addons= on a raspberry pi with HassOS, and yesterday I got some new xiaomi devices in the mailbox, started last night with adding a couple and as soon as i see a new device popping up in the GUI the deconz addon shows the error log below and then shuts down. Below is from a aqara motion sensor.

I've tried adding sensors through the phoscon gui and the deconz configuration tool, same issue with both. I thought my raspberry had a bad memory card so i swapped it out for a new one, reinstalled hassOS and installed the deconz addon again and performed a restore trough phoscon GUI. Problem still exists. 

I dont have access to the deconz GUI since I'm using the addon for hass.io, but please let me know what I can do to help you understand the error, do you need more logs, the database or anything else just let me know. 

> 12:52:36:077 SensorNode 99: LightLevel 99 added
> 12:52:36:078 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 001
> 12:52:36:078 ZCL attribute report 0x00158D0002B8F5F6 for cluster 0x0000, ep 0x01
> 12:52:36:078 0x00158D0002B8F5F6 extract Xiaomi special
> 12:52:36:078 	01 battery 3055 (0x0BEF)
> 12:52:36:078 	03 temperature 25 °C
> Segmentation fault (core dumped)
> 12:52:36:078 	04 unknown 424 (

martikainen87, 2018-12-28:
Tried adding a xiaomi double switch and that worked without any isses, so possibly it's something related to the second row of the error log above? I see that error coming back in the logs very frequently (once a minute it seems like)

> 13:13:03:724 Poll APS request to 0x000B57FFFEB787D3 cluster: 0x0006 dropped, values are fresh enough
> 13:13:03:865 ZCL attribute report 0x000B57FFFEF416E9 for cluster 0x0300, ep 0x01
> 13:13:03:976 ZCL attribute report 0x90FD9FFFFE4A89AA for cluster 0x0300, ep 0x01
> 13:13:04:681 ZCL attribute report 0x000B57FFFED2260C for cluster 0x0300, ep 0x01
> 13:13:04:920 ZCL attribute report 0x000B57FFFEBDEBE8 for cluster 0x0008, ep 0x01
> 13:13:05:000 ZCL attribute report 0x000B57FFFEB05282 for cluster 0x0008, ep 0x01
> 13:13:05:048 ZCL attribute report 0x000B57FFFEB2BFF0 for cluster 0x0008, ep 0x01
> 13:13:05:080 ZCL attribute report 0x000B57FFFEAFD780 for cluster 0x0008, ep 0x01
> 13:13:07:252 DB save zll database items 0x00000881
> 13:13:07:331 DB saved in 78 ms
> 13:13:07:331 don't close database yet, keep open for 900 seconds
> 13:13:08:042 ZCL attribute report 0x000B57FFFEB1DF7E for cluster 0x0008, ep 0x01
> 13:13:09:099 ZCL attribute report 0x000B57FFFEAFD780 for cluster 0x0006, ep 0x01
> 13:13:09:130 ZCL attribute report 0x000B57FFFEBDEBE8 for cluster 0x0006, ep 0x01
> 13:13:09:210 ZCL attribute report 0x000B57FFFEBCC491 for cluster 0x0006, ep 0x01
> 13:13:09:259 ZCL attribute report 0x000B57FFFED2260C for cluster 0x0006, ep 0x01
> 13:13:09:290 ZCL attribute report 0x000B57FFFEB2BFF0 for cluster 0x0006, ep 0x01
> 13:13:14:020 sensor 45 (lumi.sensor_motion.aq2): disable presence
> 13:13:14:024 discard sensor state push for 45: state/lastupdated (already pushed)
> 13:13:14:268 CTRL skip polling while permit join is set
> 13:13:19:720 poll node 90:fd:9f:ff:fe:f5:d8:5e-01
> 13:13:19:725 Poll APS request to 0x90FD9FFFFEF5D85E cluster: 0x0006 dropped, values are fresh enough
> 13:13:20:257 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
> 13:13:20:257 ZCL attribute report 0x00158D0001DDAA5B for cluster 0x0400, ep 0x01
> 13:13:20:261 discard sensor state push for 26: state/lastupdated (already pushed)
> 13:13:20:262 discard sensor state push for 26: state/lux (already pushed)
> 13:13:20:289 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
> 13:13:20:290 ZCL attribute report 0x00158D0001DDAA5B for cluster 0x0406, ep 0x01
> 13:13:20:293 discard sensor state push for 45: state/lastupdated (already pushed)
> 13:13:24:720 poll node d0:cf:5e:ff:fe:30:6a:51-01
> 13:13:24:723 Poll APS request to 0xD0CF5EFFFE306A51 cluster: 0x0006 dropped, values are fresh enough
> 13:13:26:120 sensor 98 (lumi.sensor_motion.aq2): disable presence
> 13:13:26:124 discard sensor state push for 98: state/lastupdated (already pushed)
> 13:13:26:677 ZCL attribute report 0x000B57FFFEBED2A6 for cluster 0x0008, ep 0x01
> 13:13:28:085 ZCL attribute report 0x00158D0002B8F5F6 for cluster 0x0400, ep 0x01
> 13:13:28:116 ZCL attribute report 0x00158D0002B8F5F6 for cluster 0x0406, ep 0x01
> 13:13:28:630 ZCL attribute report 0x90FD9FFFFEF5D85E for cluster 0x0008, ep 0x01
> 13:13:28:711 ZCL attribute report 0x90FD9FFFFEF5D85E for cluster 0x0300, ep 0x01
> 13:13:29:349 CTRL skip polling while permit join is set
> 13:13:29:415 ZCL attribute report 0x90FD9FFFFE777A8E for cluster 0x0008, ep 0x01
> 13:13:29:446 ZCL attribute report 0x90FD9FFFFE777A8E for cluster 0x0300, ep 0x01
> 13:13:30:663 0x90FD9FFFFEDD8A74 level 255 --> 254
> 13:13:30:663 ZCL attribute report 0x90FD9FFFFEDD8A74 for cluster 0x0008, ep 0x01
> 13:13:31:576 ZCL attribute report 0x90FD9FFFFEDD8A74 for cluster 0x0300, ep 0x01
> 13:13:33:720 poll node d0:cf:5e:ff:fe:f5:ff:0f-01
> 13:13:33:723 Poll APS request to 0xD0CF5EFFFEF5FF0F cluster: 0x0006 dropped, values are fresh enough
> 13:13:38:721 poll node 90:fd:9f:ff:fe:dc:cf:8e-01
> 13:13:38:729 Poll APS request to 0x90FD9FFFFEDCCF8E cluster: 0x0006 dropped, values are fresh enough
> 13:13:40:015 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
> 13:13:40:015 ZCL attribute report 0x00158D0001DDCD5B for cluster 0x0400, ep 0x01
> 13:13:40:019 discard sensor state push for 56: state/lastupdated (already pushed)
> 13:13:40:021 discard sensor state push for 56: state/lux (already pushed)
> 13:13:40:046 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
> 13:13:40:046 ZCL attribute report 0x00158D0001DDCD5B for cluster 0x0406, ep 0x01
> 13:13:40:050 discard sensor state push for 58: state/lastupdated (already pushed)
> 13:13:43:712 no button map for: lumi.sensor_motion ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
> 13:13:43:712 ZCL attribute report 0x00158D00016C11AB for cluster 0x0406, ep 0x01
> 13:13:43:715 discard sensor state push for 15: state/lastupdated (already pushed)
> 13:13:43:720 poll node d0:cf:5e:ff:fe:14:5a:22-01
> 13:13:43:724 Poll APS request to 0xD0CF5EFFFE145A22 cluster: 0x0006 dropped, values are fresh enough
> 13:13:44:448 CTRL skip polling while permit join is set
> 13:13:48:720 poll node 90:fd:9f:ff:fe:dd:8a:74-01
> 13:13:48:722 Poll APS request to 0x90FD9FFFFEDD8A74 cluster: 0x0006 dropped, values are fresh enough
> 13:13:49:444 ZCL attribute report 0x000B57FFFEBC15E3 for cluster 0x0300, ep 0x01
> 13:13:50:004 ZCL attribute report 0xD0CF5EFFFEF5FF0F for cluster 0x0300, ep 0x01
> 13:13:51:188 ZCL attribute report 0xD0CF5EFFFE306A51 for cluster 0x0008, ep 0x01
[... cut ...]
> 13:14:10:961 ZCL attribute report 0x90FD9FFFFE4A89AA for cluster 0x0008, ep 0x01
> 13:14:12:433 no button map for: lumi.weather ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 001
> 13:14:12:433 ZCL attribute report 0x00158D0001D7072A for cluster 0x0000, ep 0x01
> 13:14:12:434 0x00158D0001D7072A extract Xiaomi special
> 13:14:12:434 	01 battery 2895 (0x0B4F)
> 13:14:12:434 	04 unknown 17320 (0x43A8)
> 13:14:12:434 	05 RSSI dB (?) 7 (0x0007)
> 13:14:12:434 	06 LQI (?) 4294967296 (0x0100000000)
> 13:14:12:434 	64 temperature 2281

manup, 2018-12-30:
Interesting, can you please provide the core dump to track down where the crash happens?

The Wiki has a good description how to generate a core dump.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Enable-core-dumps-on-Raspbian