dresden-elektronik/deconz-rest-plugin #1073: unable to pair osram lightify plug, touchlinks works

malger, 2018-12-28:
I am unable to discover the osram lightify plug via the discovery function (open network)
Touchlink reset however works

logs
```
00:23:03:331 Device TTL 1941 s
00:23:08:337 dev /dev/ttyAMA0
00:23:08:339 GW firmware version: 0x262f0500
00:23:08:339 GW firmware version is up to date: 0x262f0500
00:23:42:337 discovery updated announce interval to 10 minutes
00:23:45:764 send permit join, duration: 59
00:23:47:263 DB save zll database items 0x00000008
00:23:47:325 DB saved in 61 ms
00:23:47:325 don't close database yet, keep open for 900 seconds
00:23:55:201 Device protocol version: 0x0109
00:23:55:342 Current channel 20
00:23:55:386 CTRL ANT_CTRL 0x03
00:23:55:729 Device protocol version: 0x0109
00:23:55:734 Beacon src: 0x6385 ch: 20 updateId: 3
00:23:55:734 * node has higher updateId should be 2, TODO handle
00:23:55:751 Beacon src: 0xB0C4 ch: 20 updateId: 3
00:23:55:751 * node has higher updateId should be 2, TODO handle
00:23:55:793 Current channel 20

```

manup, 2018-12-28:
While light search is running..
To reset the plug you need to press the button on the plug until you can hear a click sound.

malger, 2018-12-29:
the reset works but even afterwards the plug is not found by deconz

malger, 2018-12-29:
it works now, resetting the gateway and deleting the database solved the problem.

closed 2018-12-29 by malger