dresden-elektronik/deconz-rest-plugin #1075: Time controlled sensor

alex0411, 2018-12-29:
Hello,
is it possible to create time controlled schedules for the motion sensor?
Background: With Hue (Labs) it was possible to add different schedules for the motion sensor, e.g. from monday to friday use this motion sensor setting: 100% from 7:00 to 22:00, in the night just 20%, and on the weekend 100% from 9:00 to 0:00.

manup, 2018-12-29:
This is currently not possible with the Phoscon App, but will be during 2019.

Therefore it can only be done with custom rules via REST-API.

ebaauw, 2018-12-29:
You might want to check what the rules the Hue lab routine creates on the Hue bridge and create similar rules for deCONZ.

alex0411, 2019-04-29:
> This is currently not possible with the Phoscon App, but will be during 2019.
> 
> Therefore it can only be done with custom rules via REST-API.

Are there any updates so far? Thanks :-)

closed 2019-09-03 by stale[bot]