dresden-elektronik/deconz-rest-plugin #1077: Heiman Carbon Monoxide Sensor cannot connect

Thoit, 2018-12-31:


closed 2019-05-07 by stale[bot]

robrea, 2019-05-30:
Hi @Thoit,

Did you find a solution? I am having the same problem.

Thanks,
Rob

Thoit, 2019-05-31:
Unfortunately not @robrea never received any suggestions from Dresden Electronics

robroll, 2019-06-04:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1563

GitHubxtsnl, 2021-02-15:
I'm having the same issue.
I'm running Home Assistant 2021.2.3
deCONZ 6.7.0
ConBee II 26660700 / 2.09.03 / 12/23/2020

I've got this CO device from Ali:
https://www.aliexpress.com/item/4001031100655.html?spm=a2g0s.9042311.0.0.6da04c4dTuT4Ht

[image]

[image]

[image]

I can see it is connected, the green line, but I cannot change the Node name nor does it come up in HA.
I think it is not supported yet by deCONZ / ConBee

any ideas or suggestions on how to get this device working?

SwoopX, 2021-02-15:
The device is not supported yet. It just looks like a Heiman, but to 99% isn't one. Please raise a device support request.

GitHubxtsnl, 2021-02-16:
thank you @SwoopX for the reply. Sorry for my ignorance but how do I raise a device support request?