dresden-elektronik/deconz-rest-plugin #1078: [2.05.52] cannot pair anything at all

aep, 2018-12-31:
Hi,
it's unclear how to use the phoscon app or the deconz gui. I'd appreciate a pointer to some sort of documentation as well. i tried several devices which are supposed to be supported, but can't get any of them to join.

- hue dimmer switch
- xiaomi motion sensor
- tradfri lamp
- hue lamp
- osram plug

firmware is 2.05.52. This is a newly purchased usb thing, that i updated before doing anything else (might have been a mistake?)

i tried completely reset devices as well as joining an established network, as well as touchlink. Tried both old and new webgui. Nothing seems to work at all. The dimmer switch doesn't even indicate the xbee stick is doing anything at all.

aep, 2018-12-31:
update: touchlink actually does work. it finds one of the 5 devices near it. however, the only follow up action is then the "Reset" button, which just says "busy try again in a few seconds" 
[image]

Smanar, 2019-01-01:
Lol, less than 24h to make a try.
I m curious, wich one procedure have you using for the xiaomi motion sensor for exemple ? First, you can't do it with the old webgui.

manup, 2019-01-01:
Your ZigBee connection wasn't working (channel 0) as shown in the screenshot. Therefore the ZigBee settings needed to be checked.

Anyway happy new year..

aep, 2019-01-01:
thanks, happy new year to you as well :)

would have been nice if there was a manual for the next person (setting the channel , having to use a usb cable, etc, how to find touchlink ).

probably after i changed the channel somehow -  i still don't know how i did that -   
the xiaomi window and motion sensors could be paired by pressing the button 3 times and then scanning for new sensors in the online pwa thing. They appear fine in the pwa, but not in openhab (master build from yesterday night).

The hue dimmer appeared as node in deconz randomly long after i gave up, but it doesnt appear in the pwa ui. After factory resetting it, i could not pair it again, so not sure how it was paired in the first place.

The tradfri stuff later paired just fine. but was really slow to react compared to the original bridge (like 1-2 seconds), so not usable unfortunately.  Hue and osram never worked.

Happy to try a different firmware if there's one.

/b/

Smanar, 2019-01-01:
On my side.
For xioami sensor :  PWA / sensors / others / press 10s
For osram plug : light / press button up to listen to a 'clic'

Smanar, 2019-01-01:
And I don't understand where you have find your pairing procedure, but they are same for all Zigbee projects and written on the manual gived with device.

Even zigbee2mqtt you have talked about use same procedure > https://koenkk.github.io/zigbee2mqtt/getting_started/pairing_devices.html