dresden-elektronik/deconz-rest-plugin #1080: No database file (zll.db) found

rstoermer, 2019-01-02:
Hi,

I got Deconz running in a Docker container on a RPI2. Functionality wise everything works good, although I had two crashes two days ago (not sure if caused by the RPI, Deconz or something else), which led me to look into the syslog, where I saw the following message every 10 seconds all day. Any idea what might cause this?

> Dec 31 06:27:27 pi2 deCONZ-update2.sh[297]: no database file (zll.db) found
> Dec 31 06:27:37 pi2 deCONZ-update2.sh[297]: no database file (zll.db) found
> Dec 31 06:27:47 pi2 deCONZ-update2.sh[297]: no database file (zll.db) found

manup, 2019-01-02:
This is the update script looking for the database in specific places. These likely don't exist in the Docker container.

Is this the the docker image from @marthoc https://github.com/marthoc/docker-deconz ?

I thought the deCONZ systemd services and hence the update script don't run in this container.

rstoermer, 2019-01-02:
Yes, that is the container that I am using. I just remembered though that I first installed Deconz without the container, but then uninstalled it again to switch to Docker. Maybe something went wrong and this is the old install? If so, how can I check this?

marthoc, 2019-01-02:
@rstoermer When you say you looked "into the syslog", what do you mean? The only place that the deCONZ Docker container will show logs is by use of the `docker log [container-name]` command. If you are using `journalctl` to view systemd logs or checking under e.g. /var/log, then you will be looking at the logs of the standalone deCONZ you installed earlier, which is likely not working since you deleted the relevant directories when uninstalling. If you do `sudo systemctl status deconz` does the deCONZ service show as running? (If so, stop it: `sudo systemctl stop deconz`, and disable it: `sudo systemctl disable deconz`.

@manup You are not wrong - the systemd services do not run in the Docker container, only the main deconz application in minimal mode.

manup, 2019-01-02:
>@manup You are not wrong - the systemd services do not run in the Docker container, only the main deconz application in minimal mode.

I've checked the postinst script, by installing the deCONZ package the `deconz-update` systemd service is started. This service didn't handle the docker specifics well. For example the zll.db file wasn't found.

These issues are fixed now (2.05.53) and the database in `/root/.local/share/dresden-elektronik/deCONZ` can be found. However I'm not sure if updating deCONZ via this service is desired or if the usual "docker way" is the better approach. If you like I can prevent enabling the deconz-update service when deCONZ is installed inside a docker container.

closed 2019-05-09 by stale[bot]

hgw77, 2019-06-27:
I guess the Problem is not fixed for Raspian.

I run into the same problem because, deconz.service is not running with the root user so the update script cannot find the database :-/

On Raspian, after I enabled the deconz service the script `/etc/systemd/system/multi-user.target.wants/deconz.service` has in line 7 the entry `User=1000` so the service is running with normal user rights (in my case the first user that I created on the system) but
`/etc/systemd/system/multi-user.target.wants/deconz-update.service ` is running as root. Thats the reason why the datbase is not found.