dresden-elektronik/deconz-rest-plugin #1082: Add support for Lupus Mains Socket

tseroo, 2019-01-03:
Any chance to get this device integrated? Struggling to get the powermeter data.

[image]

[image]
[image]

[image]

tseroo, 2019-01-08:
Seems the Lupus device is identical in construction with the Bitron 902010/25, still I'm not able to get the power measurements as there is no Sensor created for the device. Any chance to get this added?

manup, 2019-01-08:
The are both branded versions of the Climax Plug. I'll add the modelid to whitelist,  the related sensors should then be created.

tseroo, 2019-01-09:
Sounds good. Just tell me if/when I should do some testing.

tseroo, 2019-01-14:
@manup Any idea when you can make the change?

manup, 2019-01-14:
Should work with 2.05.55, but the plug needs to be joined again to create the resource.

tseroo, 2019-02-02:
Finally I had some time to test this and the good news is, it is working. But some tweaking is still needed as the scaling is not correct. It reports 377 W instead of 37.7W. So both values, power and consumption, have to be divided by 10 to deliver the correct values. Thanks so far!

tseroo, 2019-02-14:
@manup Any chance you can get the scaling fixed for the Lupus plug?

closed 2019-06-21 by stale[bot]