dresden-elektronik/deconz-rest-plugin #1083: lumi.light.aqcn02

cnrd, 2019-01-03:
Hi

I just got one of the Aqara Zigbee bulbs in the door.
They are white spectrum bulbs (2700-6500k).

However they also report them selves as color bulbs it seems (at least they show up as color in home assistant).

[image]
[image]

Are there any of these clusters that are of interest?

ebaauw, 2019-01-04:
The device type indicates _Color Dimmable Light_ instead of _Color Temperature Light_.  In the _Color Control_ cluster, it should report its colour capabilities.  I would expect this to be only colour temperature (ct) for a white spectrum bulb.  If so, the rest api should not expose `state.xy` nor `state.hue`, nor `state.sat`.

ebaauw, 2019-01-05:
That is the same for the Philips Hue color temperature lights and does not indicate whether the bulb accepts a _Move to color_ command (setting xy).  Can you scroll down and check attribute 0x400A?

cnrd, 2019-01-10:
Everything except 0x400B and 0x400C is greyed out after clicking "read".

Calling the Move to color command returns UNSUP_CLUSER_COMMAND.

cnrd, 2019-01-25:
@manup any chance this is fixable?

manup, 2019-01-26:
>We need to whitelist the light to expose it as we would a ZLL colour temperature light.

Agree, it never gets boring with Aqara :)

I think these are the same lights https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1057

I'll put something up for 2.05.58

manup, 2019-01-26:
Closing this issue for now, please continue discussion in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1057

closed 2019-01-26 by manup