dresden-elektronik/deconz-rest-plugin #1084: Switch wxkg01lm

fonzies83, 2019-01-03:
Dear,
I try to add this switch using the instruction to press the reset button for 5 second after click add switch. In the Deconz the sensor is present liche lumi.sensor and if i click the button of the switch in the deconz the led of lumi.sensor blink. But in the phoscon app appear always failed to connect. In this way I cannot see the switch in home assistant and use it. Any solution?
Thank you and regards

[image]
[image]

heitec11, 2019-01-04:
Try to use at Deconz/Edit "Reset seleceted nodes" or "Delet seleceted nodes". I had the same problem in the past with another sensor.

fonzies83, 2019-01-08:
Sorry, I try several time to delete or reset the node but always the same, is showed in deconz but not in the phosocon app and home assistant. Any other solution?
Thank you.

closed 2019-05-15 by stale[bot]