dresden-elektronik/deconz-rest-plugin #1087: RaspBee Update not working

hermannstemberger, 2019-01-04:
Hi,
I'm struggling on a recently bought RaspBee Premium to get it running with HASSIO in the end.
Since it didn't work out after many hours of trying, I used your prebuilt image on a Raspberry Pi 3 B 
Raspbian Stretch RaspBee (Beta) Version: | 10-2018 | Released: | 2018-10-30

I tried to update the RaspBee module to the lastest software via terminal, but somehow I can't, can you please help me?

I run (see also on screenshot attached):
`pi@phoscon:~ $ sudo systemctl stop deconz`
`pi@phoscon:~ $ sudo GCFFlasher -l`
`GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10`
`RaspBee: /dev/ttyS0 (default)`
` `
`no FTDI devices found`
`pi@phoscon:~ $ sudo GCFFlasher -d 0 -f Downloads/deCONZ_Rpi_0x26300500.bin.GCF`
`GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10`
`using firmware file: Downloads/deCONZ_Rpi_0x26300500.bin.GCF` 
`reset via FTDI`
`FTDI can't open device 0, status = 2`
`pi@phoscon:~ $`

I did not hurt the module and use the latest downloads available... Please help me.
[image]

[image]

twiddern, 2019-01-06:
Hello,

just drop the **-d 0** statement which is intended to update the USB ConBee stick/FTDI. Since you've a RaspBee module it will then use the default /dev/ttyS0

manup, 2019-01-06:
>just drop the -d 0 statement which is intended to update the USB ConBee stick/FTDI. Since you've a RaspBee module it will then use the default /dev/ttyS0

Correct. Further details for updating RaspBee and ConBee are in the Wiki:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

closed 2019-01-06 by hermannstemberger