dresden-elektronik/deconz-rest-plugin #1089: Can't pair IKEA GU10

faxioman, 2019-01-05:
I'm unable to pair a new IKEA TRADFRI LED GU10.
I'm aware of other issue like this ... but none of the suggested solutions are working.
My conbee device has the 262F0500 version firmware.
I have already try to reset the tradfri with no luck.
I've no problems with others ikea light bulb.

faxioman, 2019-01-05:
Working after updating the deconz package (raspbian) from 2.05.48 to 2.05.54

faxioman, 2019-01-05:
... but the pair procedure is ... crazy! In order to add four of these bulbs I need to reset bulb/try pair/reset conbee/reset bulb/retry/ ... and so on ... and, of course, a single bulb at a time :(

faxioman, 2019-01-05:
Set the web ui to search before starting the reset procedure did the trick!!
Thanks @clockworkant !!!!!!!!!

closed 2019-01-05 by faxioman

RikardLegge, 2019-10-27:
For anyone still having issues with factory resetting the plain white GU10 bulbs, this tip helped me 

> If you have problems resetting the plain white (non-colour) IKEA bulb, 
> try making the "ons" very brief (less than a second) and the "offs" longer

Source: [Factory-reset-an-IKEA-bulb](https://github.com/mozilla-iot/wiki/wiki/HOWTO:-Factory-reset-an-IKEA-bulb)

basvankuijck, 2019-12-31:
A small reminder regarding this issue:
- Start the light search in the Phoscon web UI
- Switch it on/off 6 times like shown here: https://www.youtube.com/watch?v=mJm9YpPrGzk
- Finally leave the light on and wait a couple of seconds

Phoscon / deconz should pick it up and dim it slightly, then after a few seconds later the light will appear in the web ui

mountainsandcode, 2020-01-12:
Was struggling with these bulbs as well and can confirm this helps. Contrary to all other Tradfri bulbs, these need very short on periods to be reset, to the point where in many cases, the light itself doesn't even turn on before its being switched off again.

> For anyone still having issues with factory resetting the plain white GU10 bulbs, this tip helped me
>
>
> Source: [...]

HeavensBlade, 2020-03-01:
Hope this helps someone in the future:

> If you have problems resetting the plain white (non-colour) IKEA bulb,
> try making the "ons" very brief (less than a second) and the "offs" longer

New model GU10 LED1837R5 (Case completely made of glass). 
Bulb identifies as TRADFRI bulb GU10 **WW** 400lm (old model only had one W -> ...GU10 **W** 400lm)
But still I did not manage to connect it.

What helped me was a trick found on [reddit](https://www.reddit.com/r/tradfri/comments/dm0rv3/psa_moving_tradfri_gu10_bulbls_from_hue_hub_to/)

> Turn on the power to the GU10 bulb.
> 
> Hold the Hue remote adjacent to the GU10 bulb to be reset.
> 
> Simultaneously press and hold the 0 and 1 buttons on the remote. The led on the remote will flash green once.
> 
> After approx 10 seconds the led on the remote will flash orange 2 or 3 times and then go green. The bulb will then pulse dim to bright(ish).
> 
> Wait for the bulb to stop pulsing (approx ten seconds).
> 
> Turn off the bulb for 5 seconds.
> 
> **Repeat steps 4-8 (not step 9)**. No idea why but the bulbs will not be detected until the steps are repeated.