dresden-elektronik/deconz-rest-plugin #1090: Zigbee onoff controlller added as light instead of switch

clockworkant, 2019-01-05:
I'm attempting to add a zigbee onoff controller

[image]

When adding through the web ui as a switch the device is added to deconz as a light.

When I view the light in the deconz software I see the following information

[image]

I've tried turning the light off but the power is still being fed to the output. I verified that the onOff value is reporting as false but sadly it's just not working

[image]

Any thoughts on how to register the device as a zigbee on off switch?

Fl0ca, 2019-01-05:
I think this is normal, Osram Smartplugs are also added as lights. Switches are only for remotes etc.

clockworkant, 2019-01-07:
I believe something else is wrong here. The relay appears to be functional but the device is not working in deConz. I believe it is functional inside of iobroker, https://github.com/ioBroker/ioBroker.zigbee/wiki/Supported-devices where it's called "KS-SM001 (Lamp_01)"

I don't mind it looking like a light if it works. But currently it doesnt work!

Is there a way to interogate the node in deconz-gui and somehow manually trigger the relay?

manup, 2019-01-07:
>Is there a way to interogate the node in deconz-gui and somehow manually trigger the relay?

You can select the On/Off cluster and in the Cluster Info Panel trigger all commands manually.

clockworkant, 2019-01-07:
Thanks @manup, 

I tried executing the on and off comands from the on off cluster. It changes the value of the onOff boolean state (0x0000) but the device still supplies power.

When the device has been reset and is pairing it pulses on and off and I can see the output device being turned on and off and so I'm sure the relay is working.

At a loss for where to go from here!

manup, 2019-01-07:
>At a loss for where to go from here!

Good question the On/Off cluster is the only control interface the device offers. If that doesn't work there is little we can do. Does the device has a load connected?

clockworkant, 2019-01-08:
Is there any chance that the device is being mislabeled in deconz and being reported as a light instead of a switch?

I've found information on zigbee2mqtt which makes me think the switch should work https://github.com/Koenkk/zigbee2mqtt/blob/91bbb0d4a800a6d5ac6cf166fea4d53783a3e724/docs/integration/home_assistant.md

The instructions say that in order to add it to "smart things" you need to install the smart things alexa skill.

What I'm thinking is that the alexa skill is providing more device capabilities than what the device broadcasts on the zigbee network. This is just me guessing though!!

ebaauw, 2019-01-08:
It seems to be mislabeled as _Dimmable_ light, but that’s unrelated to it being exosed under `/lights`.  Wired switches and plugin units, used to control dumb lights, behave like smart lights and are exposed as such.  Wireless switches, used to control smart lights, are exposed as switches.

Smanar, 2019-01-08:
But why it's mislabeled as Dimmable light ? There is nothing in cluster or information to explain that ?
Something hardcoded in plugin ?

ebaauw, 2019-01-08:
> But why it's mislabeled as Dimmable light ? There is nothing in cluster or information to explain that ? Something hardcoded in plugin ?

No, it's hardcoded in the device firmware: the device type is in the simple descriptor of each endpoint.

ebaauw, 2019-01-08:
Hm, it might be an issue with deCONZ.  There's one entry for `Dimmable light` (with lowercase `L`) in general.xml, for the ZLL profile and device ID 0x0100.  However, this is a ZHA endpoint, and device ID 0x0100 for the ZHA profile is `On/off light` (which would be the correct device type for this device).  @manup?

There's also an entry `Dimmable Light` (with uppercase `L`) for device ID 0x0101, but I think the deCONZ GUI is case sensitive here.

clockworkant, 2019-01-08:
@Smanar the identify doesn't work however when I put the device into pairing mode (by reseting it), it flashes the device
https://www.dropbox.com/s/2tmg7c6uxixx180/Reset.MOV?dl=0

manup, 2019-01-09:
I'll check the general.xml file, but other than the label it should have no effect on functionality.

Again the question: Does the device have a connected load which it can control?

clockworkant, 2019-01-09:
@manup I'm really sorry, I'm not sure what you mean by "connected load it can control?"

manup, 2019-01-09:
I mean is the controller actually connected to anything (output) :)

clockworkant, 2019-01-09:
@manup Ah! Yes. It does have an output attached. I've also verified that during the pairing/reset state when the relay is blinking the output device is also flashing. You can see the setup in this video https://www.dropbox.com/s/2tmg7c6uxixx180/Reset.MOV?dl=0

It's why I'm so baffled that the commands aren't working. Clearly the relay is capable of switching but the deconz commands are not effecting it

manup, 2019-01-15:
Can you please also double check that the input L (phase, brown) and N (neutral, blue) as well as the output L/N are wired correctly.

clockworkant, 2019-01-15:
@manup I've rewired the plug and taken some video to show whats going on

https://imgur.com/a/JoAK0dG

When the power is toggled from deconz the output still output power however the led is changing strength slightly as you can see from the video.

I've included a photo of the wiring as well as how I am testing it in the deconz-gui

manup, 2019-01-15:
Wiring looks good now. I honestly can tell why the device doesn't electrically switch on/off. From a Zigbee perspective all is working.

One more test out curiosity, can you please press the "All off" button in the Phoscon App. The device should stay on.

clockworkant, 2019-01-15:
The physical device remains but in the web app it shows it to be off.

pponce, 2019-01-29:
I just ordered 2 of these from aliexpress.
They are very economical. Hope we can figure this out.
Will report back on my experience once i receive them and attempt to get them working.

manup, 2019-01-29:
I've also ordered one but it hasn't arrived yet.

forast3r, 2019-01-29:
I also ordered one from China, I'll report results when it arrives

forast3r, 2019-01-29:
@clockworkant did you ever try to pair it with a hue bridge?

clockworkant, 2019-01-30:
@forast3r I'm afraid I no longer own a hue bridge so I didn't attempt to pair it with the bridge.

I'm really interested to see if you guys can get it working. It feels safer than using my modified sonoff basics!

manup, 2019-01-30:
Just checked my amazon order status: delivery 7. Febr. - 19. Febr.
Will test the device as soon as it's delivered.

ademobile, 2019-02-14:
Just noticed this thread, and I am using these. Got mine of eBay (UK) and noticed I can get from Amazon (UK as well). I paired it and it works as expected. I can control from Phoscon and API with no issues. The only things I have noticed in Phoscon is no firmware version or a correct name for the item:

[image]
[image]

What would be nice if you could assign the correct icon to items (and UK version rather than euro etc.) so it makes more sense rather than it being defaulted for you!

clockworkant, 2019-02-15:
@ademobile when you mention you've got it working can you confirm that you are able to turn off a device connected to the zigbee onoff controller? I could turn mine off in the UI however the connected device would not alter it's state. If yours is working then it would mean that my device is defective!

ademobile, 2019-02-15:
From Phoscon, I can turn off or on as it is connected to my water feature as shown above, also I control it through the API via node red with no issues either

mredz, 2019-02-15:
[https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1257](url) - I started another issue here without seeing this.

manup, 2019-02-16:
Mine also arrived joined quickly and works as expected.

I've connected a E27 bulb to the controller and on/off, toggle commands do switch the relay and power properly.

[image]

The version info is fixed with latest commit, the device only support the datecode attribute.

mredz, 2019-02-16:
Thanks :-) Is this applied to a build cycle yet? If not is there instructions on how to compile on WIndows?

manup, 2019-02-16:
No and sadly no (compile instructions only for Linux) ;)

Will be part of 2.05.60 arriving during the week.

clockworkant, 2019-02-21:
@manup I have the same version as you (I read the value using deCONZ gui). I suspect my unit must be defective in someway. Thanks, I'll query it with the seller

pponce, 2019-03-01:
Just chiming in that the two zigbee on/off controllers that i ordered arrived and work as expected for me.

mredz, 2019-03-01:
> 
> 
> Just chiming in that the two zigbee on/off controllers that i ordered arrived and work as expected for me.

pponce: Are you using version: V2_05_59?

pponce, 2019-03-01:
Version 2.05.59 / 2/14/2019
Firmware 26300500

mredz, 2019-03-02:
Ok I just upgraded the firmware to 26300500 but still get the device turning on and off multiple times. I wonder if this will be fixed by the new version of deconz or if I have a faulty switch?

JaumeMuste, 2019-03-02:
I confirm that my unit with 2.05.59 is working properly with deconz.

mrdago, 2019-03-24:
I've just upgraded deConz to the latest version to check if the Zigbee OnOff Controller may work with this version. 
Version 2.05.60 / 14.2.2019
Firmware 26300500
The switch is still a "light" (that is really not important), but the device behaviour is unfortunately as described previously by clockworkant. An on or off command by the Phoscon app leads to toggling the relay of the controller. That's to bad as the device is cheap and seems to have a good manufactured quality.

manup, 2019-03-24:
>An on or off command by the Phoscon app leads to toggling the relay of the controller.

That should be expected behavior?  deCONZ can't do much more here other than sending on/off to the device. On my unit it works without issues.

Hard to tell why but it seems that some units are faulty for some reason.

andyjenkinson, 2019-04-06:
Just wanted to add, I ordered some of these from aliexpress, but the seller contacted me and asked me to cancel the order, saying that there was a quality problem. I asked for more information because I had already ordered another from a different seller [which arrived yesterday] and they said "nothing to worry about, just a problem with the manufacturer".

clockworkant, 2019-04-09:
@andyjenkinson @manup I've received some more components to play with and used the zigbee2mqtt project in order to test the switch I have. My switch is defective. I'm so sorry! I'm also upset and annoyed that I wasted £10 on it. I feel like buying ikea Tradfri plugs is a better investment.

I'm happy to close this issue.

closed 2019-08-15 by stale[bot]

Smanar, 2019-09-07:
Just for information this device have "Unknow" as type.

`{"etag":"f5f520d04118333c0edf4263018d3cde","hascolor":false,"manufacturername":"eWeLink","modelid":"SA-003-Zigbee","name":"Udelys","state":{"alert":"none","on":true,"reachable":true},"swversion":null,"type":"Unknown","uniqueid":"00:12:4b:00:1d:43:5d:d7-01"}`

uboot21, 2019-11-27:
Ok, ich habe zwei Kommentare zu diesem Topic, da ich selber stundenlang probiert habe.
1. Der Switch konnte bei mir mit allen Mitteln nicht gefunden werden. Ich habe dann irgendwann die Phase in den anderen Eingang umgetauscht und er war sofort da.
2. Ich habe das gleiche Phänomen wie oben beschrieben gehabt, das nach dem Schalten In der GUI das Licht anfing zu blinken. Dies war vorbei als ich die Lampe einem Raum zugeordnet hatte. Danach ließ sich alles wie gewünscht schalten.

muchasuerte, 2019-12-15:
I have 3 of these device they are working pretty well, just the inclusion was a bit tricky, 
is there a way to make them reported with type "On/Off plug-in unit" instead of "type": "Unknown" ? 

I'm using it to manage some devices but on UI these are reported as bulb.

Smanar, 2019-12-15:
Yes, it's possible, and "easy" but there is a problem.
Lot of manufacturer make this kind of device, so you can have different manufacture or modelid for "same device"
I can explain you where to make change in the api code if you want to make tries ?

muchasuerte, 2019-12-15:
yes, please would be very helpful being able to apply small fix

Smanar, 2019-12-15:
Lol, nope you haven't understand ^^, I can explain how to do, but I can't do it. You have the hardware, will be faster for you for test.
On the github homepage you have the procedure to do it.

File light_node.cpp in the function void LightNode::setHaEndpoint(const deCONZ::SimpleDescriptor &endpoint)
I see 2 corrections possible for this problem :
- With setting a deviceId = DEV_ID_ZLL_ONOFF_PLUGIN_UNIT
- If don't work (It depend of endpoint for exemple) make like for the "Range extender"
According to the manufacture, and if you can the modelid (but I can be missing on this part)

You are perhpas in the DEV_ID_ZLL_DIMMABLE_LIGHT situation, this id have been disabled on the code.

muchasuerte, 2019-12-15:
yes, I have just seen the code is not too tricky, the best way should be compile the code with some debug string, is it possible?  

then should I exclude and include again the device or is enough replace the library and restart deconz?

muchasuerte, 2019-12-15:
ok, I was able to build the plugin, but how can I identify the right deviceId by using the deCONZ UI?

ZCL Version: 1
Manufacturer Name: eWeLink
Model Identifier: SA-003-Zigbee
Manufacturer Code: 0x0000 (VENDOR_NONE)

but where I can find the deviceid?

Smanar, 2019-12-16:
Yes, you can compile it as it and use debug mode, to use debug, just run it with command line https://github.com/dresden-elektronik/deconz-rest-plugin/issues/55

Yes, you need to exclude and reinclude the device because this part is only triggered during inclusion. If it's realy a probleme, you can make a "fake inclusion" with putting phoscon in inclusion mode and using some fonction on deconz (I don't remember all of them, I can search if you need them) but I have less succed with this method.

And you have find a problem ^^. There is no way to find the device ID.
On deconz on the node, in the title bar you have for exemple "Home automation" = HA and "Color temperature light" = Model id.
But it's an interpretation from deconz, I haven't found a way to have the numeric value too.

On the screen the model id is "dimmable light" on deconz it's for that I m thinking it's something like 
DEV_ID_ZLL_DIMMABLE_LIGHT (and in this situation "unknow" is something normal because it have been disabled)

muchasuerte, 2019-12-16:
maybe the problem is not in deCONZ but on the device firmware who is declaring itself as On/Off Light

what I have found strange was on the general.xml there are : 
`
		<device id="0100" name="On/Off Light" icon="dev-on-off-light.png">
			<description></description>
		</device>
		<device id="0101" name="Dimmable Light" icon="dev-on-off-light.png">
			<description></description>
		</device>` 

but on the code there is: 
de_web_plugin_private.h:#define DEV_ID_ONOFF_SWITCH                 0x0000 // On/Off switch
de_web_plugin_private.h:#define DEV_ID_ZLL_ONOFF_LIGHT              0x0000 // On/Off light`
de_web_plugin_private.h:#define DEV_ID_HA_ONOFF_LIGHT               0x0100 // On/Off light
de_web_plugin_private.h:#define DEV_ID_ZLL_DIMMABLE_LIGHT           0x0100 // Dimmable light
de_web_plugin_private.h:#define DEV_ID_HA_DIMMABLE_LIGHT            0x0101 // Dimmable light `
which id should I consider the one from de_web_plugin_private or the one from general.xml?

by checking on this file ( https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/zigbee-switch.src/zigbee-switch.groovy )
this device is reported as `fingerprint profileId: "0104", inClusters: "0000, 0003, 0004, 0005, 0006", outClusters: "0000", manufacturer: "eWeLink", model: "SA-003-Zigbee", deviceJoinName: "eWeLink SmartPlug (SA-003)", ocfDeviceType: "oic.d.smartplug"`

so if I check on general.xml under profile id="0104"  maybe I should define the device 0000 like  
<device id="0x0000" name="On/off plug-in unit" description=""></device>

however for now I don't wont remove them from the wall, also I have not clear the logic I don't want play without know the deCONZ logic, for now is easier get a new device just for test meanwhile I'll write a python class to make a workaround on HomeAssistant side

Smanar, 2019-12-17:
But you want to change something on deconz or on the API ?

Profile id = 104 is normal it's HA profile ID
For the same device on generic you have
deviceId: "0103" > DEV_ID_HA_ONOFF_LIGHT_SWITCH
deviceId: "010A" > DEV_ID_Z30_ONOFF_PLUGIN_UNIT
But theses 2 devices id seem working in the code, for me you have a DEV_ID_ZLL_DIMMABLE_LIGHT

To make test it's easy, just remove the commend on line 509.

On general.xml, I m agree with you device id="0100" and like you said on deconz code it's DEV_ID_ZLL_DIMMABLE_LIGHT too
`#define DEV_ID_ZLL_DIMMABLE_LIGHT           0x0100 // Dimmable light`

So for me it can work if you re-enable the DEV_ID_ZLL_DIMMABLE_LIGHT but ONLY for your manufacturer or device id if you can
Or easier, make like the "range extender" as you have too the IDENTIFY_CLUSTER_ID cluster
```

                    if (manufacturerCode() == XXXX && deviceId == DEV_ID_ZLL_DIMMABLE_LIGHT )
                    {
                        ltype = QLatin1String("That you want");
                    }
```

muchasuerte, 2019-12-18:
I have discovered that in my case the device should be the DEV_ID_MAINS_POWER_OUTLET so should be easy fix it, I should just add this device on the switch

muchasuerte, 2019-12-18:
ok, now it works

[image]
 
with just one line I was lucky
[image]

Smanar, 2019-12-18:
Lol, seriously ? so I was totally wrong.
And "on/off plug in unit" is something normal for DEV_ID_MAINS_POWER_OUTLET finally, but how deconz can use "dimmable light" for that" ^^.
Now we know the problem it seem so easy.

Make a pull request for it, others users will also be able to use the correction too.

muchasuerte, 2019-12-18:
done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2224

muchasuerte, 2020-01-11:
@Smanar I got a sonoff BASICZBR3 which is really similar to SA-003-Zigbee also the pair procedure is really the same, now it is recognized as On/Off plug-in which is fine.

Smanar, 2020-01-11:
So your Pull is usefull too for this device.