dresden-elektronik/deconz-rest-plugin #1093: Web App - where is it?

Ltek, 2019-01-05:
After updating the firmware I can no longer get to the Web App, even when clikcing the button on deConz. It always takes me to Phoscon.  I liked the dashboard of the Web App, it was much nicer and easier to use/see all groups on the same page.

thx

wvuyk, 2019-01-06:
Manup

Am I right assuming that once the old webapp is gone we can reset Philips bulbs only with the Hue dimmer? The touchlink will not appear in Phoscon?

manup, 2019-01-06:
The old webapp will stay for a while, there are no plans yet to remove it and Touchlink is one reason for that. What will be removed or disabled in the mid future is the ability to add/configure switches in the old webapp, since this causes some issues with the Phoscon App.

>The touchlink will not appear in Phoscon?

Maybe, but currently there are no plans.

Note that the feature is also available via REST-API and could be used by third party apps.

http://dresden-elektronik.github.io/deconz-rest-doc/touchlink/

wvuyk, 2019-01-06:
@manup 
Yes, maybe you are right there. Had it for Philips Hue bridges in the plugin, but never worked on that for deCONZ. But with bulbs resetting without remotes it might be better to have it in place... Already users buy a Philips bulb, but do not have the Hue dimmer with it.

Thanks!

ebaauw, 2019-01-06:
Note that you now need to power cycle the Hue bulbs, for touchlink to work.  Since the Nov 2017 firmware, they only respond within 30 minutes after powering on.

wvuyk, 2019-01-06:
That is no problem, I can include that in instructions for the plugin. Thanks for reminding me :-)

closed 2019-05-14 by stale[bot]