dresden-elektronik/deconz-rest-plugin #1095: New Tradfri repeater

jo-me, 2019-01-06:
Hi
Ikea has listed a new Tradfri device that enhances the signal. See here https://m2.ikea.com/de/de/p/tradfri-signalverstaerker-10400408/

Sounds interesting for the price.
The effectiveness has to be evaluated though..

00lex, 2019-01-07:
you can use a bulb or a socket for the same price as extender

ebaauw, 2019-01-12:
I think they created it for their smart blinds.

AntiHeld889, 2019-02-10:
[image]
[image]
[image]
[image]
[image]
[image]

ebaauw, 2019-02-10:
Cool.  I don't think there's anything useful to expose through the REST API?

Did you buy this one separately, or did it come with the smart blinds?

AntiHeld889, 2019-02-10:
I have buy it separately, i don‘t think there‘s anything usefull for Rest API too

szoller, 2019-02-14:
@AntiHeld889 
How did you pair your tradfri repeater? mines aren't visible in DeConz (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1258 )

AntiHeld889, 2019-02-14:
Start light or sensor search and nearby reset the repeater

szoller, 2019-02-14:
I guess with reset you mean pressing the reset-button for 5 sec?

Thank you, i'll try again tonight

smingoli, 2019-03-01:
Went to Ikea today to take a look at a new desk and saw the repeater..... didn't even know it came out so I forgot the desk and grabbed one of those instead..... Seems to work like a charm...

SeraphimSerapis, 2019-03-25:
A little late to the party. Is this only for the deconz application or also for the Phoscon web interface? I did the reset, ran both sensor and light search and the repeater does not appear.

I tried with two devices to make sure it's not a hardware defect.

ebaauw, 2019-03-25:
It does (or rather should) show in the deCONZ GUI, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1258#issuecomment-464484232.  It doesn't have any functionality to to expose, so no REST API resource is created for it.  Consequently, it's invisible to apps (incl. Phoscon).

SeraphimSerapis, 2019-04-26:
I understand. A number of people -- including myself -- run deCONZ headless. That way I don't have access to the GUI and can't check if it paired correctly. Maybe some of this information could be exposed without a fully fledged implementation in Phoscon?

manup, 2019-04-26:
Jep, I think it makes sense to show it in the app perhaps with a icon and `reachable` status. I've put it on the todo list.

Dinth, 2019-04-27:
Im also waiting for it. As @SeraphimSerapis said above, i run deconz in docker in HassIO - so ive got no access to Deconz other than through Phoscon app

gekomax, 2019-05-24:
Hi, any news here?
I Need to add a Ikea pure reapeter on a conbee device that Is working on Hassio. 

It Is possibile to add the reapeter by mounting the conbee hardware on a pc, using deconz desktop software to add that Ikea reapeter and then export It to the Hassio configuration?
Thanks and Sorry for my english.
Bye

fbrinker, 2019-07-02:
This would be great.

Any chance to check, on my remote headless instance, if it paired successfully without any indicator in phoscon?

Nastras, 2019-08-09:
Is it foreseeable when it will be implemented?

pattyland, 2019-09-15:
@manup I looked through the commits from V2_05_67 and V2_05_69 and I didn't see a hint to the repeater. I think I paired mine successfully but as I don't see any feedback it's hard to tell

alundstroem, 2019-09-29:
Yeah, I agree with @pattyland, +1 on that (:

manup, 2019-09-29:
Sorry took a while, it is added now as light and will be part of the next release.
The led on the repeater can be used to identify the device via the `light.state.alert` attribute.

[image]

alundstroem, 2019-09-29:
Sounds awesome, will be looking forward to upgrading then (;

pattyland, 2019-09-30:
@manup Thanks! Do I have to repair the repeater to see it in Phoscon? Or should this work automatically?

If anyone is interested, this is the commit: 627a742de94ad2021133d946c79a013f2b885b08
I would like to be able to implement this for #1899 too :(

alundstroem, 2019-09-30:
@pattyland, +1 on both questions/suggestions (;

manup, 2019-09-30:
>@manup Thanks! Do I have to repair the repeater to see it in Phoscon? Or should this work automatically?

Fortunately repairing isn't needed.

* Start the lights search
* Power-cycle the repeater

The repeater should then be added to the list of lights.

NedSin, 2019-10-04:
Hey together, 
i run Deconz Version 2.05.69 AddOn on HassOS with RaspBee Module.
I`m happy my new motion Sensor and on/off switch from ikea is directly working i just dont get this repeater to work. 
Try recycle the power, press the reset for 5 seconds. I`m not 1m away from the RaspBee Module. 
My english is not the best is this repeater in Phoscon Version 2.05.69 now Supported or do we HASS people has to wait for next release ? 

Thanks for the Answer 

Best Greetings 
Ned

manup, 2019-10-04:
The repeater will be part of the next version 2.05.70, which should arrive soon.

NedSin, 2019-10-04:
Thank you for the Info, also for the great Work ...

tnowak, 2019-10-15:
I'm running version 2.05.67, firmware 264A0700 now and I'm totally unable to add Ikea repeater as a light now (as on screenshot above). I've tried power cycling, reseting (5s), etc.

Do I need to wait for 2.05.70 to get it in Phoscon app?
Any news when 2.05.70 will be released?

pattyland, 2019-10-17:
@tnowak As @manup wrote 2 comments before yours the repeater will be part of 2.05.70. If you look at the release cycle I guess there will be one in the next 2 weeks.

Tip: If you see the repeater in deCONZ it does its job. The listing in Phoscon doesn't matter

tnowak, 2019-10-17:
I have Conbee stick in docker on RPi, I suppose deCONZ app is for Win/USB only (?)

pattyland, 2019-10-18:
@tnowak This is no the right place to ask for general support. I'm running deCONZ in Docker on a RPi too and use VNC to connect to the container and use the App. This has nothing to do with your client system

tnowak, 2019-10-18:
@pattyland thanks for the tip. I've connected with VNC and found... mess of blocks and lines :) After moving those blocks with mouse I've figured out that it presents a diagram of connections. Ikea reperater is actually there, unfortunately it seems it does not do repeater job (connected only with powered devices). I've bought it to extend range to Aqara_Door_Sensor_2 exclusively and it doesn't :/
[image]

LintHart, 2019-10-24:
@tnowak I read somewhere on my initial research into deCONZ that xiaomi sensors always look like they’re coming directly off the main gateway, regardless of where they’re actually routing through. I could be very wrong and have no evidence to support but if I find the thread I’ll link it. 

I’ve just started the migration process myself from a xiaomi gateway to a Conbee 2 stick so I’d be very interested in getting the ikea repeater if it all works together. 

Just to confirm, the graph shows one thing, but does the door sensor happen to function reliably? Do you think that with the distance of the sensor it may actually be routing through a repeater somewhere?

Cheers,

Linton

tnowak, 2019-10-24:
@LintHart Didn't seem to me. I've noticed no distance extension on that Aqara Door Sensor neither when Ikea Repeater was pluged-in somewhere in between nor when plugged closer the and of ZigBee range. I ditched it for now and moved to another technology with that sensor.

LintHart, 2019-10-24:
@tnowak how interesting. Perhaps the xiaomi stuff simply does not play well with others (which wouldn’t surprise me in the slightest). Out of interest for my own future, what technology did you move to with that sensor?

tnowak, 2019-10-24:
@LintHart RF 433Mhz, but let's stop here as this is out of topic of this issue

LintHart, 2019-10-25:
This may help someone in the future, I plan to try it myself. Basically it seems if you add a repeater after your Xiaomi sensors are set up, they will not automatically attach themselves Like other devices. We need to add the repeaters and then re-pair the sensors in the location they’re to be used so they latch on to the appropriate repeater. 

Advice seems to be repair and rename to previously used to avoid mishaps in things like home assistant

More discussion here:
https://forums.homeseer.com/forum/lighting-primary-technology-plug-ins/lighting-primary-technology-discussion/jowihue-w-vuyk/1300637-zigbee-extender

closed 2019-11-19 by manup

S474N, 2020-06-20:
> > @manup Thanks! Do I have to repair the repeater to see it in Phoscon? Or should this work automatically?
> 
> Fortunately repairing isn't needed.
> 
> * Start the lights search
> * Power-cycle the repeater
> 
> The repeater should then be added to the list of lights.

This is only for light or also for Ikea Tradfri Repeater? I cannot add it :(

Smanar, 2020-06-20:
The ikea repeater haven't a small hole and you need to use a paperclip to press the buton ?

S474N, 2020-06-20:
> The ikea repeater haven't a small hole and you need to use a paperclip to press the buton ?

Thank you! Solved! :)))

Range extender is in lights :)

adrianmihalko, 2020-07-11:
Previously I had successfully connected Tradfri Repeater, but after upgrade it is not connected:

[image]

Do I need to re-join every device again after fw upgrade?

Smanar, 2020-07-11:
No, perhaps it need a power cycle, off/off. But no, router ususaly don't need to be re-included (it s another story for Xiaomi devices)