dresden-elektronik/deconz-rest-plugin #1097: Incorrect value for lastinstall date in swupdate2

Thomas-Vos, 2019-01-07:
The REST API returns an incorrect value for the `lastinstall` date. I received multiple reports for this in my app Hue Essentials and noticed it myself as well when I updated to 2.05.54. (previous version I had installed was 2.05.48).

GET request URL: /api/myusername/config
Part of response:
``` json
{
    "swupdate2": {
        "autoinstall": {
            "on": false,
            "updatetime": ""
        },
        "bridge": {
            "lastinstall": "%cI",
            "state": "allreadytoinstall"
        },
        "checkforupdate": false,
        "install": false,
        "lastchange": "",
        "lastinstall": "",
        "state": "allreadytoinstall"
    }
}
```

KodeCR, 2019-01-07:
The lastinstall field used to be empty, I changed it to report the last git commit date to make the Hue App happy, i.e.:

GIT_COMMIT_DATE = $$system("git show -s --format=%cI HEAD")

in de_web.pro (that's also where the time field formatting comes from).

That works for me on a raspberry pi also. Did those users compile the rest plugin themselves?

Thomas-Vos, 2019-01-07:
This is the response I received after updating. I did not compile the rest plugin myself, I just started the update using the Phoscon web app. deCONZ is installed on a Raspberry Pi.

bdickhaus, 2019-01-09:
I can confirm this on two Raspberry Pi's with RaspBee cards. I am doing an install, not a compile. A third install is returning a very abbreviated response that does not even include this information.

KodeCR, 2019-01-10:
Ok, the released package indeed has this behaviour. If you build the plugin yourself (and have git) it's fine.
@manup Does the machine where the release package is built handle "git show -s --format=%cI HEAD" properly?

manup, 2019-01-17:
The hue specific attributes update attributes are not completely implemented, since for historical reasons we've used a own mechanism.

Here is how the Phoscon App handles updates via /config endpoint given that `Accept: vnd.ddel.v1` is set for all requests:

*  check that `fwupdatestate` is `idle`, meaning no firmware update is running
* `swversion` is the current version
* `swupdate.version` is the latest update version, might be the same as `swversion`

When a new version is available it can be updated by:

    POST http://<gateway-ip>/api/<apikey>/config/update

While update is running we poll `swupdate2.state` and check for `installing` and `transferring` to indicate the update is busy. Note that also a timeout can occur while the software is restarting.

Thomas-Vos, 2019-01-19:
Thank you very much for explaining, @manup. The firmware/software update process is clear to me now. The next version of Hue Essentials will have this implemented (as well as some other stuff that was not yet made deCONZ compatible). Just have a few more questions (and bugs?) about the updates:

- I noticed when doing a POST request to `/config/update`, and then checking the `/config` response, sometimes `fwneedupdate` reports true (the Phoscon app then also shows a firmware update). I think this is a bug because there is no firmware update available. (`fwversion` and `fwversionupdate` are both `0x262f0500`):
```
	"fwneedupdate": true,
	"fwupdatestate": "idle",
	"fwversion": "0x262f0500",
	"fwversionupdate": "0x262f0500",
```

- Is there a way to check for updates using the REST API? Just like in the Hue API you can set `swupdate2.checkforupdate` to true. I couldn't find anything for this in the source code.

- Is `updatechannel` implemented fully? I tried setting this to `alpha`/`beta` but am not able to get the latest update (2.05.55) through the REST API. I know I can install it manually, but am wondering if there is a way to get this (beta?) version through the API.

Thomas-Vos, 2019-01-20:
Can confirm this is fixed in version 2.05.55.

closed 2019-01-20 by Thomas-Vos

manup, 2019-02-16:
> I noticed when doing a POST request to `/config/update`, and then checking the `/config` response, sometimes `fwneedupdate` reports true (the Phoscon app then also shows a firmware update). I think [...]

I've notices this too. https://github.com/dresden-elektronik/deconz-rest-plugin/commit/97cb7bfd0ea2aba9ab6596c03928d32cd4ffaca3 is related to this, there will be more fixes for the update feature in next versions.

>* Is there a way to check for updates using the REST API? Just like in the Hue API you can set `swupdate2.checkforupdate` to true. I couldn't find anything for this in the source code.

Currently the check is done automatically by the discovery service. The manual check via `swupdate2.checkforupdate` will be added shortly.

>* Is `updatechannel` implemented fully? I tried setting this to `alpha`/`beta` but am not able to get the latest update (2.05.55) through the REST API. I know I can install it manually, but am wondering if there is a way to get this (beta?) version through the API.

Kind of, the channels exist since 2013 :) there is a difference between stable and beta, currently alpha is the same as beta. I'm planning to support separate alpha channel soon.

Some side notes on the current update channels:

* **stable** is way behind, 2.04.35, this will be updated in a few weeks
* **beta** is updated more often, currently 2.05.46, but isn't synced with the latest version released on GitHub. Here the plan is to sync it with recent GitHub version a few days after release
* **alpha** will be synced with GitHub version