dresden-elektronik/deconz-rest-plugin #1099: Phoscon App changed values in timing

Ronoon, 2019-01-07:
Since the update to  .. 54 the timing for light Control has changed.
Before i could use 1 or 2 min for Lights off now it is up to min 3 minutes on Raspberry 
an the odd it is on PC up to 4 minutes :-(.
Looks like the window scaling fails to show the uppervalues in the box because off the screen values.
Maybe you should configure the timing values as input values and not as defined boxes ( or maybe the boxes should be scrolling to the upper values).

Screenshot from PC values:

 
[image]

manup, 2019-01-07:
>Looks like the window scaling fails to show the uppervalues in the box because off the screen values.

Good catch, seems indeed to be the problem. Looking forward to fix this in 2.05.55.

Ronoon, 2019-01-07:
Well thx

Ronoon, 2019-01-22:
Well we are at .57 and this issue hasn´t been fixed. 
Maybe it´s just forgotten to fix due to the workload you have there.

manup, 2019-01-22:
It was already fixed in the last version. Maybe your Browser cache hasn't updated yet.

You can force update the cache with a random query string in the url bar like:

    http://<your-gateway-ip-address>/pwa/group-sensors.html?12345

[image]

Ronoon, 2019-01-22:
Yep 
had to reinstall the complete precompiled version to run it the you showed.
The simple cache cleaning wouldn´t help with this issue.
But thx for the reply, it is really appreciated

closed 2019-01-22 by Ronoon