dresden-elektronik/deconz-rest-plugin #1100: Device intergration Bitron AV2010/21B window sensor with ext.contact

Ronoon, 2019-01-07:
[image]
Could you pls integrate the window sensor 

[image]
[image]
[image]
[image]

Ronoon, 2019-01-13:
Integration is about 50% done cause only the "Namespace" is recognized but there is something missing in this part or somewhere else:

                case IAS_ZONE_CLUSTER_ID:
                {
                    if (modelId.startsWith(QLatin1String("CO_")))                     // Heiman CO sensor
                    {
                        fpCarbonMonoxideSensor.inClusters.push_back(ci->id());
                    }
                    else if (modelId.startsWith(QLatin1String("DOOR_")))              // **Heiman door/window sensor**
                    {
                        fpOpenCloseSensor.inClusters.push_back(ci->id());
                    }
                    else if (modelId.startsWith(QLatin1String("PIR_")) ||             // Heiman motion sensor
                             modelId.startsWith(QLatin1String("902010/22")))          // Bitron motion sensor
                    {
                        fpPresenceSensor.inClusters.push_back(ci->id());
                    }
                    else if (modelId.startsWith(QLatin1String("GAS_")) ||             // Heiman gas sensor
                             modelId.startsWith(QLatin1String("SMOK_")) ||            // Heiman fire sensor
                             modelId.startsWith(QLatin1String("902010/24")) ||        // Bitron smoke detector
                             modelId.startsWith(QLatin1String("lumi.sensor_smoke")))  // Xiaomi Mi smoke sensor

Missing are the integration for 902010/21 +/32

In the commit of Ma Ca was this originally integratet but there is also nothing anymore

Ronoon, 2019-01-14:
Just got the including done so the 902010/21 will show up in the API and Tada its working
Just added the line for the 902010/21 into the de_web_plugin.cpp as shown below:

case IAS_ZONE_CLUSTER_ID:
                {
                    if (modelId.startsWith(QLatin1String("CO_")))                     // Heiman CO sensor
                    {
                        fpCarbonMonoxideSensor.inClusters.push_back(ci->id());
                    }
                    else if (modelId.startsWith(QLatin1String("DOOR_")) ||	      // Heiman door/window sensor
			     modelId.startsWith(QLatin1String("902010/21")))          // Bitron window sensor
                    {
                        fpOpenCloseSensor.inClusters.push_back(ci->id());
                    }
                    else if (modelId.startsWith(QLatin1String("PIR_")) ||             // Heiman motion sensor
                             modelId.startsWith(QLatin1String("902010/22")))          // Bitron motion sensor

manup, 2019-01-18:
Thanks for figuring this out, I've added this to the repository.

Ronoon, 2019-01-22:
Well thx for the integration of this sensor, maybe you could integrate it into the visible part of Phoscon so that it could be used through the app.
Also the is this little issue with the timing problem about the sliders in the app.....

manup, 2019-02-02:
The sensor should be visible in 2.05.58, can you please check if it works?

closed 2019-06-09 by stale[bot]