dresden-elektronik/deconz-rest-plugin #1101: Deconz tries to open on port 80, conflicts with web server.

Andreas-Pedersen, 2019-01-07:
Windows 10 Pro
Installed Deconz 2.0.54, and tried to open webapp and Phoscon App but 
since i already have a webserver running on port 80(Homeseer), this doesnt work.
Are there any workarounds?

-Andreas
[image]
[image]

Smanar, 2019-01-07:
There is a command line to change server port 
`--http-port=<port>`

But I haven't tried it on windows.

Andreas-Pedersen, 2019-01-07:
Worked perfect! Changed the Deconz shortcut. Added the 
[image]

> --http-port=8080

closed 2019-01-07 by Andreas-Pedersen

pillemats, 2021-01-06:
works not for me... any other solution ?