dresden-elektronik/deconz-rest-plugin #1102: Phoscon app - finds indefinate GW's

Andreas-Pedersen, 2019-01-07:
Why does this happen?

[image]

johol, 2019-01-08:
After updating to 2.05.54 I have the same problem and can not log in.

johol, 2019-01-08:
Downgrading to 2.05.52 fixed the problem for me.

johol, 2019-03-13:
Today I did a update to 2.05.60. The problem still exists.

johol, 2019-03-15:
It seems to be a problem with the bridgeid but I found a ugly workaround.
Use sqlite3 command to find your bridgeid
sqlite3 zll.db "select value from config2 where key=\"bridgeid\""
00212EFFFF00B79A

After this put the id in the login.html page

--- login.html.orig     2019-03-15 12:56:02.712971203 +0100
+++ login.html  2019-03-15 14:11:17.158575108 +0100
@@ -267,6 +267,9 @@
                     var rsp = xhr.response;
                     rsp.id = rsp.bridgeid;
                     rsp.ip = url.hostname;
+                    if (typeof rsp.id==="undefined") {
+                         rsp.id = '00212EFFFF00B79A';
+                     }
                     if (url.port)
                         rsp.port = url.port;
                     rsp.reachable = true;

johol, 2019-03-18:
Today I tried some older git commits. With commit e7cb33f0f61d801f33400cb0f037cb06a7060427 title "Add authorisation status to request and move rest handlers back to rest_configuration."  the problem started.

johol, 2019-07-18:
The problem still exists in 2.05.66. Before commit e7cb33f0f61d801f33400cb0f037cb06a7060427 login is working as expected.

closed 2019-11-22 by stale[bot]