dresden-elektronik/deconz-rest-plugin #1104: "hue" or "sat" values set via restAPI are not retained

asgothian, 2019-01-08:
I have an issue when setting hue and sat values on a Philips Hue color light, using the restAPI in connection with iobroker.deconz:

iobroker sends a well formed setLightState with only the sat or hue state and its new target value
restAPI respons with status ok, and confirms the value in the response 
The attached lamp changes color / saturation.
No Events for changed values are recorded at iobroker.deconz (neither for hue, sat nor xy)
iobroker requests the light state via getLightState
restAPI responds with the full light state, but the values for hue, sat and xy are the values which were present before the setLightState with the changed hue or sat state was sent, essentially reporting a wrong color back to iobroker.

Hardware : Raspberry Pi with RaspBee, deconz 2.05.52, latest RaspBee firmware (Phoscon does not report a new firmware present)

ebaauw, 2019-01-08:
I think this is the same issue as #1096.

Looks like the REST API plugin updates its cache of the light state when setting `state.hue` (note the changed `colormode` and `xy` values), but not correctly (`hue` is not updated, and the `xy` values are wrong).  Does deCONZ take into account the gamut supported by the light when computing `xy` from `hue`/`sat`?

Would it be an idea to fast-poll the light after setting its color (and the member lights when updating a group's color)?  That way, the REST API plugin wouldn't have to guess the color.

In homebridge-hue, I don't use `state.hue` nor `state.sat` (except for lights that don't support setting `state.xy`) and translate the HomeKit Hue and Saturation values (in RGB color space) to `xy` (in CIE 1931), taking into account the supported color range (gamut) of the light.
```
$ ph get /lights/221
{
  "ctmax": 500,
  "ctmin": 153,
  "etag": "52723b013e5bacb234489b60b34e85fe",
  "hascolor": true,
  "manufacturername": "Philips",
  "modelid": "LCT003",
  "name": "Living Room 1",
  "state": {
    "alert": "none",
    "bri": 254,
    "colormode": "xy",
    "ct": 459,
    "effect": "none",
    "hue": 13074,
    "on": true,
    "reachable": true,
    "sat": 205,
    "xy": [
      0.510593,
      0.414988
    ]
  },
  "swversion": "5.127.1.26420",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:00:f0:71:18-0b"
}
$ ph put /lights/221/state '{"hue": 0}'
$ # light has turned red
$ ph get /lights/221/state
{
  "alert": "none",
  "bri": 254,
  "colormode": "hs",
  "ct": 459,
  "effect": "none",
  "hue": 13074,
  "on": true,
  "reachable": true,
  "sat": 205,
  "xy": [
    0.596042,
    0.329815
  ]
}
$ # read color control cluster in deCONZ GUI
$ ph get /lights/221/state
{
  "alert": "none",
  "bri": 254,
  "colormode": "hs",
  "ct": 153,
  "effect": "none",
  "hue": 0,
  "on": true,
  "reachable": true,
  "sat": 205,
  "xy": [
    0.620582,
    0.333875
  ]
}
```

asgothian, 2019-01-22:
Fixed for me with 2.05.57

closed 2019-01-22 by asgothian