dresden-elektronik/deconz-rest-plugin #1105: Ikea bulb and driver DeConZ 2.05.54 seems offline?

PrimaryInc, 2019-01-08:
Hi,

Dont know it this is something others using Ikea bulbs with decent, but after upgrading to .54 on firmware 0x26300500 2 x Ikea devices have a strange status.

First light 69 - kitchen bench is a Trådfri driver (TRADFRI transformer 30W) for LED and one TRADFRI bulb GU10 W 400lm seems offline.

Now, I did check the Trådfri driver works even if status in deconz/phoscon is offline.
version .52 did not have this problem.

I have not checked the bulb yet if it works or not, but will do so later today.

[image]

manup, 2019-01-08:
They should come back when you power-cycle them. .54 should run better with Ikea since way less requests are send, which did stress them in prior versions.

PrimaryInc, 2019-01-08:
Hi @manup,

Agreed but the drives does not come back, power cycled number of times.
I haven't tried to power cycle the bulb yet.

PrimaryInc, 2019-01-08:
Quick update, the ikea bulb (light 37) came back online by it self now. Driver stil offline, but works.

1968-Larry, 2019-01-08:
Trådfri driver only works with power on/off after upgrade phoscon. App shows 30% but the lights never changes. When I use a hue android app it works setting xx% and the light changes. Perhaps deconz works but problem with phoscon? I also got the offline problem.

PrimaryInc, 2019-01-08:
Hi @1968-Larry, 

Power cycle has been done without change.. Issue seen in picture, occurred in 54 and was not an issue in 52, so not sure what your point is :)

1968-Larry, 2019-01-08:
I got the same problem + setting xx% brightness isn’t working for me after upgrading to .54.

PrimaryInc, 2019-01-08:
Ok, the funny thing is that everything works fine in my environment, just deconz that has it marked red. 
I've also updated multiple version of deconz without the need to power cycle the driver. Another strange thing I noticed is that in my homeseer environment I use jowihue plugin to control deconz bulbs and so on. After .54 in need to disable and enable the HS plugin to make HS control work after reboot of HS.

1968-Larry, 2019-01-08:
Ebaauuw: The group is off but the light is on. Only one light in the group. I change the light manually in group. The value I set doesn’t stays if I refresh the browser.

tubalainen, 2019-01-08:
Ive got three drivers (two 10W model and one 30W) and they all work fine after update to .54 .

PrimaryInc, 2019-01-08:
Thanks for feedback @tubalainen :)

manup, 2019-01-08:
When you select the red node and press `0` (zero) key does it react?

tubalainen, 2019-01-09:
> Thanks for feedback @tubalainen :)

Worth noting is that Ive got half the amount of your lights (ish). The majority are IKEA Trådfri.

PrimaryInc, 2019-01-15:
So I had a power outage at my house here the other day, After starting up everything, all bulb's status is ok. Cheer for feedback :)

closed 2019-01-15 by PrimaryInc