dresden-elektronik/deconz-rest-plugin #1107: Get Osram power plugs recognized as power plugs not light

jaal2001, 2019-01-08:
Hi,
by now I have not found a similiar request for this, so please bear with me ifm, if there already has been one.
I do have connected only Osram Power plugs to Deconz by now, but all get recognized as lights in the hub. This results seeing them in homekit/homebridge also as light.
So when I say "Siri, lights off" the power plugs are also switch off.

Is there any way to present the power plugs as power plug and not as light within Deconz?

Thanks
JJ

ebaauw, 2019-01-08:
Yes, make homebridge-hue expose them as outlets.  See https://github.com/ebaauw/homebridge-hue/releases/tag/v0.10.0.

jaal2001, 2019-01-08:
Okay, but there is no way to display them correctly in the Deconz GUI, right? I assumed Deconz would pass this information to homebridge.

ebaauw, 2019-01-08:
deCONZ exposes the device type as reported by the plug and makes this available through the RESP API to homebridge-hue.  Wired switches typically report themselves as a light instead of a plug, and some plugs simply report the wrong type, so homebridge-hue no longer uses this.  Also, there was a request to be able to expose a group (of plugs) as an _Outlet_.  Hence the decision to list the resources to be exposed as _Outlet_ explicitly in a resourcelink.

jaal2001, 2019-01-09:
Okay, understood. Migration works just fine. Thanks for your help (and all your efforts in these groups here!)

closed 2019-01-09 by jaal2001