dresden-elektronik/deconz-rest-plugin #1114: Sensors not visible in Phoscon, and unable to add sensors in 2.05.54

jeopold, 2019-01-09:
After upgrading Deconz from 2.05.34 to 2.05.54, all my sensors (Xiaomi temperature, Xiaomi door sensors, Xiaomi water sensor, Philips Hue motion) disappeared from Phoscon, but they are accessible via the Deconz GUI, and they seems to communicate normally to my Home Assistant server. 

I am also unable to add a new sensor, from Phoscon as well as from the Deconz GUI.  

I run Deconz under Ubuntu 16.04 LTS. 
My Conbee is updated to 262F0500.

manup, 2019-01-09:
Can you please press F12 while Phoscon App is open and then reload the page. Are there any Errors shown?

Maybe the devices are not shown due a Javascript error..

>and they seems to communicate normally to my Home Assistant server.

This indicates they are still there in the REST-API but for some reason are not shown.

>I am also unable to add a new sensor, from Phoscon as well as from the Deconz GUI.

It is only possible to add new devices via Phoscon App since only from there device search can be started.

jeopold, 2019-01-09:
Sorry, not really familiar with looking for errors in the browser, what should i look for and where?

Reverting to 2.05.34 did not get the functions back by the way.

manup, 2019-01-09:
I don't think it's the version, more like a cache issue in the browser.

Which browser are you using? May as a short test open the Phoscon App in a private Tab or another test (so a fresh copy is used not a cached one).

jeopold, 2019-01-09:
You were absolutely right, everything back to normal after logging out and in from Phoscon again. Thank you!

closed 2019-01-09 by jeopold

manup, 2019-01-09:
Normally this should happen automatically, I just found the bug why it didn't, will be fixed in next version.