dresden-elektronik/deconz-rest-plugin #1116: Run deCONZ on setup with Hass.IO with Raspbee or separate ConBee

corvy, 2019-01-10:
Hello, I am looking to do for instance mesh investigation and IKEA device OTA firmware upgrades but that is not possible from Phoscon on Hass.IO with Raspbee. Could I by a Conbee and join the same netowork from a PC to do upgrades and other tuning or would all devices become confused? I really does not want to rejoin my bulbs and switches.... 

Another option could be to boot Raspbian on a separate SDcard on the Raspberry Pi with the Raspbee hat. Would that work? Could I join the active network of devices in that way and run deCONZ in Raspbian? 

Thanks for any advice!

manup, 2019-01-10:
The recent VNC PR for the Docker container might be helpful: https://github.com/marthoc/docker-deconz/pull/81

>Could I by a Conbee and join the same netowork from a PC to do upgrades and other tuning or would all devices become confused? I really does not want to rejoin my bulbs and switches....

Joining is theoretically possible but would cause lots of problems since two gateways would try to conquer the network :)

>Another option could be to boot Raspbian on a separate SDcard on the Raspberry Pi with the Raspbee hat. Would that work? Could I join the active network of devices in that way and run deCONZ in Raspbian?

That works, but make sure to copy the configuration files like zll.db to the sd-card as well.

In summary I think the VNC option might be the simplest.

corvy, 2019-01-10:
VNC is great, I will test asap! Thanks 🐱

corvy, 2019-01-10:
@manup do you know how I can pull the #81 PR onto my Hass.IO setup? Is there a command-line instruction to pull directly form GitHub maybe? I have only used the marthoc add-on store from the GUI to install. Maybe I can via my host OS SSH do a manual "git pull" over the existing install? If you could point me in the right direction I can probably make it work. 

Secondly, can I add VNC settings to the Hass.IO interface directly? I will also ask the same on the #81 PR directly. Thanks!

corvy, 2019-01-10:
Well I cannot find a way, so I will wait until the PR gets merged. Does not seem to be far off. Thanks for the help!

manup, 2019-01-10:
Nope, sorry I have only very brief experience with docker. Looks like the PR will be merged soon.

closed 2019-05-18 by stale[bot]