dresden-elektronik/deconz-rest-plugin #1117: Phoscon UI Incorrect

chimpy, 2019-01-10:
Hello,

The Phoscon UI is inaccurate (not showing all my sensors), and the lights are showing inaccurate statuses, and are not responsive.

However the paired devices *are* responding and showing correct statuses inside Home Assistant.

Restarting deconz has not solved the problem.

Any ideas?

starrs-github, 2019-01-10:
Hi,
same problem here.

I‘m running raspbee with this information:

Hersteller dresden elektronik
Produkt RaspBee
Version 2.05.54 / 4.1.2019
Firmware 262F0500
Die Version ist auf dem neuesten Stand.

New paired Osram Plugs are no more working, also unpaired (and till this time working) and new paired plugs will no more work.

Also, any ideas?

tubalainen, 2019-01-10:
If you push F12 in your browser and then reload the page .. is there any websocket errors?

I made a booboo my self when setting up my docker image... Setting up the websocket port different on the "outside" and "inside" of the docker `-v 4434:443` and that caused me similar symtoms like you are describing...

chimpy, 2019-01-10:
I don’t use Docker I have the .deb installed directly on my system

chimpy, 2019-01-11:
Installing deconz-2.05.52-qt5.deb over 2.05.54 made my sensors and lights reappear. The lights are responsive again.

chimpy, 2019-01-12:
Update: The UI has now gone back to exactly as it was in 2.05.54. The devices remain responsive in Home Assistant, but not in Phoscon UI.

manup, 2019-01-12:
Are they shown as reachable in Phoscon App or are they shown inactive (light grey)?

chimpy, 2019-01-12:
@manup These are all my zigbee lights below. The lights do nothing when I click on them from the UI. The statuses are also incorrect (except for Living Room), all the lights are off.

[image]

In addition, the are a whole load of sensors missing. 6 are shown but I have 20 paired.

[image]

Again, the weird thing is that whilst the UI is unresponsive to inputs or status changes, Home Assistant *can* turn the lights on, and *is* responsive to sensor changes.

manup, 2019-01-12:
Hmm smells like Javascript error. Can you please open developer console in browser (F12) and reload the page (F5). There might be some errors  shown.

chimpy, 2019-01-12:
Normally I use Safari. I went to the page in Chrome, opened the Console and then refreshed. After waiting a while, the Console settled on "(2) manual poll". Then I tried to control Living Room by clicking on the light in the left-hand menu, this triggered the errors below:

[image]

manup, 2019-01-12:
Is it possible that you have multiple deCONZ instances running?
503 error says that Zigbee network isn't connected, but in this case also Home Assistant wouldn't work.

Is Home Assistant also using deCONZ on port 8080?

chimpy, 2019-01-12:
Here is a screenshot of a `grep -i` on `ps`. Looks like there is more than one process but I only launched it once with `sudo systemctl start deconz`.

[image]

Here is the configuration Home Assistant created automatically via discovery:

{
                "connection_class": "local_push",
                "data": {
                    "allow_clip_sensor": true,
                    "allow_deconz_groups": true,
                    "api_key": "<snip>",
                    "bridgeid": "<snip>",
                    "host": "192.168.1.26",
                    "port": 9042
                },
                "domain": "deconz",
                "entry_id": "6bbc56b1f6ad4514a3b52058a442e201",
                "source": "user",
                "title": "deCONZ-00212EFFFF0329B9",
                "version": 1
            },

manup, 2019-01-12:
Hmm this looks weird usually only one instance can be started and the default parameters look different.

What is the output of: `systemctl list-units | grep deconz`

manup, 2019-01-12:
Also with `ps axf`you can look what the parent process of the instances is in a tree structure.

chimpy, 2019-01-12:
@manup Thanks for the continued help! Here is the result of `systemctl list-units | grep deconz` and `ps axf | grep -i deconz`:

[image]

chimpy, 2019-01-13:
@manup Thank you for your patience. I have `nginx` running on the server so I've done `sudo systemctl edit deconz` to change the ports. I did the following steps:
- `sudo systemctl disable deconz-gui`
- Restarted the machine
- Home Assistant no longer responded. I assumed this was because it had decided to use port 9042. Don't know where it got that from. Changed the config to 8080 and did a `systemctl restart homeassistant`.
- Phoscon UI is now responsive and accurate.
- Home Assistant is also responsive and accurate.

Win! Seems like a PEBKAC situation. Sorry. I will wait 48 hours just in case, then close the ticket.

manup, 2019-01-13:
Cool, thanks yes this looks much better now :)