dresden-elektronik/deconz-rest-plugin #1120: All lights unreachable

MikeRVNL, 2019-01-11:
[image]
[zll.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2751247/zll.zip)

I have a situation here which looks like a stubborn one. Last week Friday I upgraded my deCONZ Windows installation to the latest version, 2.05.54. After upgrading deCONZ started normally and it ran fine during the evening.
The next morning nothing was working anymore. When I checked deCONZ, all lines between lights and sensors were gone. Looking deeper into things I noticed deCONZ status was “joining” and not in the network anymore. Also noticed the deCONZ was in Router mode and not in coordinator mode anymore. So I tried to change back to coordinator mode, pressing buttons save and done, but whatever i tried, deCONZ stays in “joining” and jumped back to router mode each time.

I then found issue #999 and applied the firmware 0x26300500 to my USB stick. This helped, I now could restart the stick and change back to coordinator mode. 
But since then all routers are unreachable. A few of my sensors (philips motion sensors) show some lines directly to my usb stick after a while, but also many (mainly XIaomi / Osram) stay behind, not showing a connection.

By the looks of it, my NWK update id has been reset. I do not know on what level it has been set in the past, but I have tried raising it not up to 8 till now, but still lights stay unreachable. What is the best way to proceed from here, I am already missing control over a week now? Is there a way to retain the lights and get them connected again?

manup, 2019-01-12:
Very likely the network parameters are messed up somehow.
Can you please send a current backup to mpi@dresden-elektronik.de I can have a look which parameters need to be set.

Please don't reset anything yet otherwise it won't help to change parameters.

manup, 2019-01-12:
Just saw the attached zll.zip ... checking

manup, 2019-01-12:
> By the looks of it, my NWK update id has been reset. I do not know on what level it has been set in the past, but I have tried raising it not up to 8 till now, but still lights stay unreachable. [...]

According to the database the update id was 0 when it was working. The parameters in the database look actually fine.

Can you please make a screenshot of the deCONZ network settings?

MikeRVNL, 2019-01-12:
ok, here it is.
[image]

MikeRVNL, 2019-01-12:
Yes, that seems the trick.. connections are being made, had to do it multiple times. Network id was automatically changed back and not accepted. after 4 times it was ok.
Thank you!

This happend overnight, how can I prevent that it forget its settings?

closed 2019-01-13 by MikeRVNL