dresden-elektronik/deconz-rest-plugin #1121: IKEA smart blinds

ebaauw, 2019-01-12:
Did any-one already get their hands on the IKEA FYRTUR and KADRILJ smart blinds?  The are controlled through the TRÅDFRI hub, so deCONZ should be able to support these.  Unfortunately, they seem to come only in grey and at fixed (non-adjustable) widths of 60, 80, 100, 120, and 140 cm.

Oliviakrkk, 2019-01-19:
I am afraid we have to wait till Feb 2...

Klagio, 2019-01-21:
In few days should hit the stores here. I will buy one for sure. If I am not mistaken they are battery operated, so end-devices

Kane610, 2019-01-21:
I'm planning on getting two on 2nd of February

ebaauw, 2019-01-21:
> There is no direct control of end-device to end-device without the repeater?

Without a router.  The repeater is definitely needed when you want to control the smart blind standalone, without a hub.  However, the coordinator is also a router, so I think/hope you can ditch the repeater if you connect both the remote and the smart blind to deCONZ (or, if possible, to an IKEA hub).

mink2k, 2019-03-03:
Ikea told us they where postponed 'till summer due to manufacturing issues and problems with the implementation to the IKEA bridge. I am also very interested in these blinds and hope they are controllable via deconz.

ebaauw, 2019-07-01:
Still waiting on IKEA...

manup, 2019-07-18:
According Reddit the release of the blinds might happen in October in the Netherlands:
https://www.reddit.com/r/tradfri/comments/c8e7ge/blinds_kadrilj_fyrtur_on_sale_in_october

Kane610, 2019-07-18:
In Sweden they will arrive within a month (hopefully)

ebaauw, 2019-07-18:
End of August in the Netherlands, see:
https://www.ikea.com/nl/nl/catalog/categories/departments/Textiles/10701/:

> Vanaf eind augustus 2019 zijn de elektrische FYRTUR en KADRILJ rolgordijnen verkrijgbaar bij IKEA. Met een afstandbediening open en sluit je eenvoudig de rolgordijnen. Ook te gebruiken in combinatie met de TRÅDFRI verbindingshub en bijbehorende app.

ebaauw, 2019-07-31:
Hm, they removed the text from the web site...

Oliviakrkk, 2019-07-31:
Looks like we need to look at Xiaomi Aqara Rolling shutter motor 
https://s.click.aliexpress.com/e/J1vGk12

elmgrenen, 2019-08-01:
Released today, hopefully this will be integrated fast. I want to buy a few :D

Oliviakrkk, 2019-08-01:
No custom sizing :/. Too bad.

Klagio, 2019-08-01:
In Italy they released the spare battery :p

CopyCat73, 2019-08-06:
Should now be available soon. They have appeared on the site in the Netherlands this week; initially it said available on august 18th. I will be buying some fyrturs.

Thoit, 2019-08-19:
I've bought them this weekend. If someone can provide me a step by step guide about what I need
to do. I will send the information.

anatolyk82, 2019-08-19:
I guess [this link](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support) is about it.

Kane610, 2019-08-19:
I also bought 2 Fyrtur, but I don't know when I will have time to play with them.

Revorge, 2019-08-19:
I've gotten two Fyrtur smart blinds today as well. Unfortunately I am running deconz headless, so haven't got access to the GUI, but if theres any other way I might help - just say the word :)

Kane610, 2019-08-19:
If you run deconz in docker you can actually enable desktop gui to be viewable over vnc

th3cube, 2019-08-19:
> No custom sizing :/. Too bad.

I think they are ;)

https://youtu.be/PL6LPZZoFlo

Oliviakrkk, 2019-08-20:
I need length 240cm. Ikea sells only 195:/.

Thoit, 2019-08-20:
I'm running Deconz as an Addon in Hassio. So I'm affraid I cannot access the gui.

Kane610, 2019-08-20:
> I'm running Deconz as an Addon in Hassio. So I'm affraid I cannot access the gui.

Yes you can; https://github.com/home-assistant/hassio-addons/tree/master/deconz#accessing-the-deconz-application-and-viewing-the-mesh-via-vnc

Thoit, 2019-08-20:
I'm able to login via VNC now. I've pressed the two buttons on the Ikea blinds (step 2). I also set permit join (255). But I don't see any changes in the VNC view. Is there something else I should do? 

ADDING DEVICES TO YOUR
WIRELESS OPEN/CLOSE REMOTE
When the wireless Open/Close remote is sold
together with a wireless roller blind and a
Signal Repeater (in the same package), they
are already paired.
To add more wireless roller blinds, just repeat
the steps below:
Make sure that your wireless roller blind is
turned on.
1 Open the back cover of the Open/Close
Remote and find the pairing button.
2 Short press on both buttons on the blind.
This will make the device awake and ready
for pairing for 2 minutes.
3 Hold the Open/Close remote very close to
the wireless roller blind you want to add: no
more than 5 cm away from .
4 Press and hold the pairing button for at
least 10 seconds on the wireless Open/
Close remote.
5 A red light will shine steadily on the Open/
Close remote. On your wireless roller blind
a white light will begin to dim and flash until
the devices have been successfully paired.
Up to 4 wireless roller blinds can be paired
with 1 Open/Close remote.

AntiHeld889, 2019-08-20:
For those who come from Berlin and are interested in a FYRTUR, they have some in Berlin Walterdorf.

[image]

Thoit, 2019-08-20:
You also can check online in what shop they are available :)

ebaauw, 2019-08-20:
Still waiting for the Fyrtur to become available in Amsterdam...

As I understand, pressing the two buttons sets the blind in touch-link pairing mode.  That’s not going to help pairing it to deCONZ.  If its like any of the Trådfri devices, resetting it to factory settings should do the trick.  For the controls, this is done by holding the pairing button for 10 seconds.

If you cannot get the blind to join the deCONZ network after a reset, you might try and pair the remote to deCONZ (again, by resetting it) and then touch-link the blind to the remote.

Thoit, 2019-08-20:
> Still waiting for the Fyrtur to become available in Amsterdam...
>
> As I understand, pressing the two buttons sets the blind in touch-link pairing mode. That’s not going to help pairing it to deCONZ. [...]

Hereby the connected remote information

[image]

Thoit, 2019-08-20:
And the information of the blinds:
[image]

Revorge, 2019-08-20:
To reset and pair the blind itself you most press both buttons on the blind for 5 seconds (not on the remote) and thus the blind should appear in the GUI as and end device - at least that's my understanding as i currently can access the GUI myself. Might reinstall my pi to desktop version to test later today...

ebaauw, 2019-08-20:
@Thoit looks like deCONZ hasn’t yet read the descriptors in full.  Can you try and read them from the _left_ of the two dropdown rounds on the node?  Make sure to wake the device first.  Also, can you please post the other info from https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support?  Please read the _Basic_ cluster attributes before making the screenshot.  Again, make sure the device is awake when reading.

Thoit, 2019-08-20:
Is this the information you need?

[image]
[image]

ebaauw, 2019-08-20:
The remote is looking good.  As I would expect, it has a client _Window Covering_ cluster, to send comands to the blind.  This means that it can only control (directly) the blind, and that the blind cannot be controlled (directly) by the other control devices.  Not too happy about the 0xFC7C cluster, which is probably used for some IKEA-specific configuration.

Can you also post the node an clusters for the blind?  And the _Basic_ cluster after reading the attributes.  The radio on the blind needs to be awake for the read to succeed; I’m not sure how to achieve that.  Hopefully the blind polls its parent every couple of seconds or it wouldn’t be able to react to the controller.

beatpaul, 2019-08-20:
Hello !
I'm happy about the progress including the ikea blinds :)
One question about the blinds: can the Endposition be customized ? --> my windows i want to use them for are less than 1,95. 
--> I couldn't find any info about that in the ikea manual

Thoit, 2019-08-20:
> Hello !
> I'm happy about the progress including the ikea blinds :)
> One question about the blinds: can the Endposition be customized ? --> my windows i want to use them for are less than 1,95.
> --> I couldn't find any info about that in the ikea manual

Yes, that's possible. Set the end position and click 2 times on one of the buttons on the blinds.

beatpaul, 2019-08-20:
> Yes, that's possible. Set the end position and click 2 times on one of the buttons on the blinds.

Thx ! .. so as soon as they're available i'll  get some :)

Kane610, 2019-08-20:
@manup will you support in getting proper Ikea cover support in deconz possible in the short term?

Thoit, 2019-08-20:
I couldn't find how to get more information about the blinds. Already pushed the 2 buttons each, simultaneously, double clicked but don't see more information then provided in the screenshot.

paolotremadio, 2019-08-21:
I have managed to get my hands of some of them today. I'm happy to provide any technical detail that can help with the implementation :)

Kane610, 2019-08-21:
@manup any progress or plan for this?

Revorge, 2019-08-23:
Today I finally paired one of my blinds to deconz, but same as Thoit I do not see how to get more information from it. I can get the node info, but there is apparantly no clusters to show or read from...

[image]
[image]

ebaauw, 2019-08-23:
deCONZ hasn’t read the descriptors.  You might try and read these manually from the (only visible) left drop-down  (on the right side of the node).  Depending on how sleepy the blind is, you might need to wake it before it responds.  Not sure how that’s done, I guess/hope by pressing a button on the blind.

Revorge, 2019-08-23:
Blind clusters & Basic cluster read
[image]
[image]

ebaauw, 2019-08-24:
Cool!  Could you try and read the _Window Covering_ cluster?  And check whether you can control the blind through its _Open_ and _Close_ commands?

Thoit, 2019-08-24:
How did you connected the blinds with Deconz @Revorge? Ik tried a couple of Times, but couldn't get it to work :(

paolotremadio, 2019-08-24:
@ebaauw here’s my multi-adapter Homebridge blind implementation. It could be useful for implementing the blinds support on Homebridge-hue
https://github.com/paolotremadio/homebridge-webshades

Which makes me wonder: should it be part of Homebridge-Hue? It is technically not hue... :)

ebaauw, 2019-08-24:
homebridge-hue has been supporting smart blinds since [v0.10.3](https://github.com/ebaauw/homebridge-hue/releases/tag/v0.10.3).

Revorge, 2019-08-24:
> Cool! Could you try and read the _Window Covering_ cluster? And check whether you can control the blind through its _Open_ and _Close_ commands?

I can control the blind from deconz GUI using the "UP / Open", "Down / Close" and "Go to lift percentage" exec buttons :)

Not sure if the attributes read completed properly, but this was all i god no matter what i tried:
[image]
[image]
[image]
[image]

> How did you connected the blinds with Deconz @Revorge? Ik tried a couple of Times, but couldn't get it to work :(

Open the network via deconz GUI or "add new lights" in phoscon and hold both buttons on the blind for about 5 seconds - the blind appeared instantly.

manup, 2019-08-24:
Looks good, I like that the blinds also provide a Poll Control cluster. The remote seems to be the same as the on/off remote that was sold before and is supported since 2.05.58.
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_58

There is sure some additional code needed, for setup a binding in order to let the remote control control the blinds directly.

As short term goal I guess it makes sense to support the blinds via the existing `/lights` hack. Not my favorite but the v2 REST-API will take some time, so lets do the hack first and add a proper blinds API in v2 later :)

ebaauw, 2019-08-24:
Yesterday I got an email notification that the 140cm FYRTUR was available at IKEA Amsterdam.  I went there straight away, but they couldn't find the FYRTURs.  As the stock had changed, I figured they found them   Made another trip this morning, but they had only some 60cm and 80cm units on left.  The remaining 140cm unit listed on the website went MIA.  They don't know when new stock will arrive.  None of the other IKEA stores within decent driving distance currently have any stock, so I guess I'll wait another week...

Anyway, whitelisted the FYRTUR (and, I hope, the KADRILJ) so the REST API plugin should create the `/lights` resource with `type` `Window covering device`.  I hope some-one can compile and test my latest commit.

Exposing the battery level will be a bit more challenging, as there's no `config` on `/lights` resources.  I could introduce a ZHABattery `/sensors` type or maybe better wait for API v2 `/devices`.

> The remote seems to be the same as the on/off remote

No, it has different firmware, a different device type, and I suspect it sends _Window Covering_ commands instead of _On/Off_ and _Level Control_ commands.

I added the remote device type to `general.xml` and whitelisted it for creating a `/sensors` resoruce.  I think that's enough, as it also advertises a client _On/Off_ cluster.  For full support, I need to see which commands it sends.  Could some-one please run deconz with `--aps-info=2` and capture the commands, including the `asdu`?

Revorge, 2019-08-24:
Really great to see that some progress and commits are being made :) If I might be of any further help in adding support - just say the word. I am currently testing all this on a "burner" SD-Card anyway, so I do not mind doing some "tinkering" :)

Thoit, 2019-08-24:
Great @ebaauw how can I test your code via the Hassio add on? Or isn't that possible?

ebaauw, 2019-08-24:
You need to get my latest commit, compile the REST API plugin and install it, see https://github.com/dresden-elektronik/deconz-rest-plugin#get-and-compile-the-plugin.

Revorge, 2019-08-24:
> You need to get my latest commit, compile the REST API plugin and install it, see https://github.com/dresden-elektronik/deconz-rest-plugin#get-and-compile-the-plugin.

I tried compiling from your repo, but I am getting an error
[image]

ebaauw, 2019-08-24:
Damn, that's a known issue with the post-v2.05.66 commits, which should be fixed in v2.05.67.  See #1732 and https://github.com/dresden-elektronik/deconz-rest-plugin/commit/1777accdc688a2a72f762f6e8d38df68dec34fef.

Revorge, 2019-08-24:
> Damn, I forgot about that. That's a known issue with the post-v2.05.66 commits, which should be fixed in v2.05.67. See #1732 and [1777acc](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/1777accdc688a2a72f762f6e8d38df68dec34fef).

That helped :)

[image]

Kane610, 2019-08-24:
Lets get this released!

Kane610, 2019-08-24:
What are the issues with the Home Assistant slider? Wrong scale?

Revorge, 2019-08-24:
> What are the issues with the Home Assistant slider? Wrong scale?

The only "problem" is that Home Assistant is completely unaware of the blind state. The slider scale is correct but Home Assistant has no clue what the current position of the blind is. You might temporarily solve this simply by having Home Assistant remember the last position set, but that is a matter for the Home Assistant community.

[336 more replies]

closed 2021-05-14 by close-issue-app[bot]