dresden-elektronik/deconz-rest-plugin #1123: 2.05.54 fw 262F0500 Unable to pair Philips Hue Lamps

asgothian, 2019-01-12:
I am running a raspberry pi with a raspbee using deconz 2.05.54 (headless) and firmware 262F050. Before the update to the latest deconz software, i was able to pair 3 Philips Hue Lamps, but since the update, i have been unable to pair new Hue lamps, despite resetting the lamps multiple times while trying to pair. The same holds true for a new Philips dimmer switch.
I have been able to pair Ikea Trädfri lamps , power drivers and remotes without issues, and control of the already paired switches and lamps works without fault.

I did double check that i can pair the affected lamps / remote to a hue bridge which i still have lying around, so there is no fault with the hue components.

What can i do to find out what is going on ?

manup, 2019-01-12:
deCONZ 2.05.54 is a bit stubborn in some cases and adding devices takes longer as needed. This will be fixed in deCONZ 2.05.55 arriving soon (hopefully tomorrow).

Meanwhile you may increase the light search duration as described here:

https://doc.phoscon.de/app/doc.html#settings-gateway-advanced-en

How do you reset the Hue bulbs?  The easiest way is with help of the dimmer switch.

asgothian, 2019-01-12:
Thank you for the quick response.
I already upped the search time to 60 minutes, but was unable to pair any hue components. I reset the bulbs with a Philips dimmer switch too.. i think i tried about 30 times.
I'll await the 2.05.55 and hope it will be less stubborn.

A.

closed 2019-05-19 by stale[bot]