dresden-elektronik/deconz-rest-plugin #1124: device support: iCasa switches

ebaauw, 2019-01-12:
[wifimedia.eu](https://www.wifimedia.eu/nl/brands/icasa/) kindly provided me with iCasa switches to integrate in deCONZ and homebridge-hue.

The battery-powered Pulse 2, Pulse 4S, and Pulse 8S keypads seem like regular switches.  They pair to deCONZ on first attempt.  They all look the same from a Zigbee perspective, except for the _Model Identifier_:
- Pulse 2: `ICZB-KPD12`;
- Pulse 4S: `ICZB-KPD14S`;
- Pulse 8S: `ICZB-KPD18S`.

[image]

deCONZ receives APS indications for the_OnOff_ cluster (0x0006) on pressing the _On_ or _Off_ button; for the _Scenes_ cluster (0x0005) on pressing or holding (store scene?) a scene button; and for the _Level control_ cluster (0x0008) on holding and on releasing the _On_ of _Off_ button(except for the 4S - the light blinks instead of burns while holding the button).  I haven't yet brought out the sniffer, but this seems pretty standard.

I can bind the _Power configuration_ cluster to the RaspBee, but get unsupported cluster commands when trying to read or write the report configuration for _Battery Voltage_ or _Battery Percentage Remaining_.  We might need to poll the switches for the battery percentage.  They seem to poll their parent after a button is pressed.

The battery percentages are 100 (3.2V), 35 (2.5V) and 98 (3.0V) - I think they report in 1%, like the IKEA Trådfri, instead of the standard 0.5%.

They appear to be OTA upgradable; the OTAU plugin shows them with _Version_ 0x0000000A and _Image_ 0x2FAA.

The _Power Descriptor_ does not seem valid.  I see ZDP answers to 0x8003, but the _Node Info panel_ continues to show them as _Mains powered_.  The _Node Descriptor_ seems fine, though.
[image]

manup, 2019-01-12:
Cool, the dimmer looks very interesting have it already on my wish list since a few days :)

> deCONZ receives APS indications for the_OnOff_ cluster (0x0006) on pressing the On or Off button; for the Scenes cluster (0x0005) on pressing or holding (store scene?) a scene button; and for the [...]

Do they send the commands without a former binding? And how, unicast or group-cast?

ebaauw, 2019-01-12:
> Do they send the commands without a former binding? And how, unicast or group-cast?

Need to sniff.  I thought this first one I tried didn't, but the other two did.
```
Jan 12 21:30:09 pi1 deCONZ[23850]: 21:30:07:510 APS-DATA.indication srcAddr: 0x000d6f0010658a6e, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -52
```
Address mode 1 is group cast, isn't it?

ebaauw, 2019-01-12:
> Cool, the dimmer looks very interesting have it already on my wish list since a few days

Here it is:
[image]
[image]

As I was advised, no metering and no reporting for the connected push switch, but attribute reporting for _OnOff_ and _Current Level_.  No _PowerOn OnOff_, but working _PowerOn Level_.  When set to 0, it doesn't light up on power cycle.  It reports 255 as group capacity, even after I joined it to a group.  The manual says it can be controlled by up to 30 Zigbee remotes or keypads, so I assume 30 groups.  OTAU seems OK, _Version_ 0x00000014 and _Image_ 0x27ED.

There's two push buttons on the device: one for reset, one for setting the minimum level.  I don't see the minimum level reported anywhere over Zigbee.  The dimmer seems small enough to hide behind a pulse switch.  At 220V it's rated for 200W for LED lamps, drivers or electronic halogen trafos; and for 400W for incandescent lights.  It supports resistive and capacitive loads - no mention of inductive (so no go for my ceiling fans).

The RESP API plugin already seems to like it well enough.  We might need to whitelist it to configure the attribute reporting.  The settings were 1, 65535 (and 0 for the _Current Level_) - I don't suppose the plugin did set these values?
```json
{
  "etag": "1c07beb922f18c4ca7cb719c4d51c5bc",
  "hascolor": false,
  "manufacturername": "icasa",
  "modelid": "ICZB-IW11D",
  "name": "Light 1",
  "state": {
    "alert": "none",
    "bri": 129,
    "on": true,
    "reachable": true
  },
  "swversion": "2.3.1_r20",
  "type": "Dimmable light",
  "uniqueid": "00:0d:6f:00:0f:94:86:ef-01"
}
```

ebaauw, 2019-01-12:
After binding the _OnOff_ and _Level_ clusters to the RaspBee and configuring reporting (1, 300, 1), deCONZ seems to receive the reports.  While transitioning from bri: 1 to bri: 254 over 600 seconds, it only sends a report per minute, though:
```
Jan 12 22:05:56 pi1 dc_eventlog[25650]: /lights/1/state: {"on":true}
Jan 12 22:05:57 pi1 dc_eventlog[25650]: /lights/1/state: {"bri":1}
Jan 12 22:07:57 pi1 dc_eventlog[25650]: /lights/1/state: {"bri":7}
Jan 12 22:08:56 pi1 dc_eventlog[25650]: /lights/1/state: {"bri":32}
Jan 12 22:09:56 pi1 dc_eventlog[25650]: /lights/1/state: {"bri":57}
Jan 12 22:10:56 pi1 dc_eventlog[25650]: /lights/1/state: {"bri":82}
```
I still need to wire a pulse switch to see how it reports external input.

manup, 2019-01-12:
Great this looks like a really nice device to fix the usual wall switch problem. Power measurement would be a nice feature but the Green Power Proxy support is also cool.

ebaauw, 2019-01-13:
> Do they send the commands without a former binding? And how, unicast or group-cast?

They send groupcasts.  They pick a random group after reset, but after binding the _OnOff_, _Level_, and _Scene_ clusters manually to another group, they use that group.  The scene commands are hard-coded for group 0x0000 and scene 0x01, ..., 0x06.  Very Hue: use the groupcast to recall the "global" scene only for the lights in the group.

I think I got them working in deCONZ, except for polling the battery percentage.  It's only populated when manually reading the _Power Configuration_ cluster from the deCONZ GUI.  @manup please help me: where do I whitelist the sensor for polling?

manup, 2019-01-13:
> They send groupcasts. They pick a random group after reset, but after binding the OnOff, Level, and Scene clusters manually to another group, they use that group. The scene commands are hard-coded [...]

Guess this is good, just like Ikea remote works with the random group. Setup is easier with this, no binding is required. 

>I think I got them working in deCONZ, except for polling the battery percentage. It's only populated when manually reading the Power Configuration cluster from the deCONZ GUI. @manup please help me: where do I whitelist the sensor for polling?

I think this needs more logic. Usually polling sleeping devices is tricky and needs to planned carefully  on  a per device basis. For Xiaomi it is done right after the special reports since this is the only time they do mac data polling.

ebaauw, 2019-01-21:
I think I managed to read the battery on receiving a message from the keypad, see above commit.  It almost seems too easy.  The keypads seem to check for firmware updates every ~10 minutes, which triggers the read, see log below.

I currently wait 300s before next reading the battery.  My test network doesn't care, but I'm not sure if this is a wise choice for real.  It might be prudent to wait longer.  The Hue dimmer switch is configured to report battery every 5m; but the Hue motion sensor only every 2h.  In order not to cause any additional traffic to the latter, we would have to wait for over 2h, which seems a bit long.  Maybe whitelisting for which devices to read the battery is a better option?  @manup, please advise.

```
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:579 APS-DATA.indication srcAddr: 0x000d6f0010658a63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0019, lqi: 255, rssi: -56
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:579 #011asdu: 013701012412aa2f0a000000010e
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:579 APS-DATA.request id: 237, addrmode: 0x03, addr: 0x000d6f0010658a63, profile: 0x0104, cluster: 0x0019, ep: 0x01 -> 0x01 queue: 0 len: 4 tx.options 0x04
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:745 read attributes of 0x000D6F0010658A63 cluster: 0x0001: [ 22:26:30:745 0x0021 22:26:30:745 ]
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:745 add task 14732 type 19 to 0x000D6F0010658A63 cluster 0x0001 req.id 238
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:745 APS-DATA.request id: 238, addrmode: 0x03, addr: 0x000d6f0010658a63, profile: 0x0104, cluster: 0x0001, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:792 Erase task req-id: 238, type: 19 zcl seqno: 62 send time 0, profileId: 0x0104, clusterId: 0x0001
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:792 APS-DATA.confirm id: 238, status: 0x00 SUCCESS 
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:792 APS-DATA.confirm request id: 238 -> erase from queue
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:30:871 aps request id: 238 finished, erase from queue
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:31:618 APS-DATA.indication srcAddr: 0x000d6f0010658a63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0001, lqi: 255, rssi: -58
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:31:618 #011asdu: 083e012100002053
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:31:619 Node data 0x000d6f0010658a63 profileId: 0x0104, clusterId: 0x0001
Jan 21 22:26:31 pi1 deCONZ[19357]: 22:26:31:619 update ZCL value 0x0001/0x0021 for 0x000D6F0010658A63 after 589 s
Jan 21 22:26:31 pi1 homebridge[2491]: #033[36m[Hue]#033[39m #033[90mPulse 8S: config changed event: {"battery":83}#033[39m
Jan 21 22:26:31 pi1 homebridge[2491]: #033[36m[Hue]#033[39m #033[90mPulse 8S: sensor battery changed from 82 to 83#033[39m
Jan 21 22:26:31 pi1 homebridge[2491]: #033[36m[Hue]#033[39m Pulse 8S: set homekit battery level from 82% to 83%
Jan 21 22:26:31 pi1 dc_eventlog[1935]: /sensors/4/config: {"battery":83}
Jan 21 22:26:35 pi1 deCONZ[19357]: 22:26:31:621 Websocket 127.0.0.1:45558 send message: {"config":{"battery":83},"e":"changed","id":"4","r":"sensors","t":"event","uniqueid":"00:0d:6f:00:10:65:8a:63-01-1000"} (ret = 119)
Jan 21 22:26:35 pi1 deCONZ[19357]: 22:26:31:625 Websocket 127.0.0.1:45928 send message: {"config":{"battery":83},"e":"changed","id":"4","r":"sensors","t":"event","uniqueid":"00:0d:6f:00:10:65:8a:63-01-1000"} (ret = 119)
```

ebaauw, 2019-01-22:
For those looking to buy an icasa Pulse keypad or the icase Dimmer: [Wifimedia](https://www.wifimedia.eu/en/brands/icasa/) give a 10% discount if you enter discount code **homebridgehue**. They ship to the EU.

Klagio, 2019-01-31:
In Home Assistant will it work as receiving events (each press will be an event)?

Klagio, 2019-01-31:
> For those looking to buy an icasa Pulse keypad or the icasa Dimmer: [Wifimedia](https://www.wifimedia.eu/en/brands/icasa/) give a 10% discount if you enter discount code **homebridgehue**. They ship to the EU.

20 Euro shipping cot ... crazy, 70 Euro for a switch ... I pass

ebaauw, 2019-01-31:
> In Home Assistant will it work as receiving events (each press will be an event)?

I don't know Home Assistant, but should be.  See my earlier posts on the buttonevent values.

ebaauw, 2019-05-24:
They now ship the iCasa dimmer with new ZigBee 3.0 firmware:
[image]

They also upgraded the ZGP function from proxy to combo.  Unfortunately, the firmware is not available for download.

They also introduced an in-wall switch (relay):
[image]

I looks, feels, and behaves very much like the dimmer.  Notably, it also supports attribute reporting, especially useful when the relay is operated thru a connected push button.  Obviously, the _Level Control_ cluster is for responding to wireless switches that send _with (On/Off)_ commands; the _Current Level_ reflects the level requested by the most recent command, but the relay on turns on or off.

[bol.com](https://www.bol.com/nl/l/icasa-domotica-systemen/N/28915/filter_N/4268562454/?view=list) now carry them as well.  I'm not sure if they'll deliver these outside the Netherlands and Belgium.

ebaauw, 2019-05-24:
Above commit adds support for the in-wall switch.  Had to do some fiddling to suppress `state.bri`, as they use VENDOR_PHILIPS as manufacturer code.
```json
{
  "etag": "53351eda0ec6d21a3bc65cc72330a073",
  "hascolor": false,
  "manufacturername": "icasa",
  "modelid": "ICZB-IW11SW",
  "name": "Light 5",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": "2.4.1_r33",
  "type": "Dimmable light",
  "uniqueid": "00:0d:6f:00:0f:d8:56:6c-01"
}
```

ebaauw, 2019-05-24:
Above commits add support for the in-wall switch.  Had to do some fiddling to suppress `state.bri`, as they use VENDOR_PHILIPS as manufacturer code.
```json
{
  "etag": "d9311f4799782dce6400dce88b742d16",
  "hascolor": false,
  "manufacturername": "icasa",
  "modelid": "ICZB-IW11SW",
  "name": "Light 5",
  "state": {
    "alert": "none",
    "on": true,
    "reachable": true
  },
  "swversion": "2.4.1_r33",
  "type": "On/Off light",
  "uniqueid": "00:0d:6f:00:0f:d8:56:6c-01"
}
```

ebaauw, 2019-05-24:
This is interesting: the Hue bridge recognises the switch as `certified` and exposes it natively to HomeKit.  Presumably because of the _Manufacturer Code_?
```
{
  "capabilities": {
    "certified": true,
    "control": {},
    "streaming": {
      "proxy": false,
      "renderer": false
    }
  },
  "config": {
    "archetype": "classicbulb",
    "direction": "omnidirectional",
    "function": "functional",
    "startup": {
      "configured": true,
      "mode": "safety"
    }
  },
  "manufacturername": "icasa",
  "modelid": "ICZB-IW11SW",
  "name": "Hue light 1",
  "productname": "Hue light",
  "state": {
    "alert": "select",
    "mode": "homeautomation",
    "on": false,
    "reachable": false
  },
  "swupdate": {
    "lastinstall": "2019-05-24T11:32:23",
    "state": "noupdates"
  },
  "swversion": "2.4.1_r33",
  "type": "On/Off light",
  "uniqueid": "00:0d:6f:00:0f:d8:56:6c-01"
}
```

manup, 2019-05-24:
>This is interesting: the Hue bridge recognises the switch as certified and exposes it natively to HomeKit. Presumably because of the Manufacturer Code?

Hehe interesting, perhaps a nice little hack,seems like it was done on purpose :) I wonder how long it will last.

ebaauw, 2019-05-24:
More fun: the switch supports _PowerOn On/Off_ and _PowerOn Level_.  The dimmer (both old ZLL and new ZigBee 3.0 firmware) supports only _PowerOn  Level_, but it won't turn on with _PowerOn Level_ set to 0.

ebaauw, 2019-05-24:
> Hehe interesting, perhaps a nice little hack

I take it you’re already cooking up a custom firmware for the FLS-PP to check this?

ebaauw, 2019-05-24:
I also works for the dimmer (at least for the ZigBee 3.0 firmware - haven't tested the ZLL firmware).  The Hue app claims power-on behaviour is not supported, though (presumably because of the missing _PowerOn On/Off_).

ebaauw, 2019-05-24:
And here's the 12V LED dimmer:
[image]

Same firmware version and functionality as the in-wall switch, except, of course, this one does dim the output.  Is already handled by the previous commits.

```json
{
  "etag": "db972e9151c3094839fbebd079b20b57",
  "hascolor": false,
  "manufacturername": "icasa",
  "modelid": "ICZB-DC11",
  "name": "Light 7",
  "state": {
    "alert": "none",
    "bri": 1,
    "on": true,
    "reachable": true
  },
  "swversion": "2.4.1_r33",
  "type": "Dimmable light",
  "uniqueid": "00:0d:6f:00:10:68:1b:d9-01"
}
```

JBS5, 2019-08-21:
I guess it is not possible to use direct bindings with the Pulse 8S like possible with the Hue Dimmer Switch and Tradfri grey/Black dimmer to keep it possible to switch a light even when deCONZ is offline?

ebaauw, 2019-08-21:
Yes it is.  You need to create a group with an ID matching the switch’s `config.group`.  I don’t think you can specify the group ID when creating a group through the REST API, though so either:
- Create a group, shutdown deCONZ and patch the group ID in the database; or
- Create a group and bind the switch to this group in the _Bind dropbox_ panel in the deCONZ GUI; or
- Add the group ID from the switch’s `config.group` to a light using the _Cluster Info_ panel in the deCONZ GUI for the light’s group cluster.

JBS5, 2019-08-21:
@ebaauw Tanks. Does this mean this kind of binding is possible with all kind of switches paired in deCONZ, for example the the Xiaomi Aqara double wirelss switch (WXKG02LM)?

ebaauw, 2019-08-21:
No, only with switches that send commands to lights or groups.  Xiaomi switches only send reports to the coordinator.

You want to look for switches that:
- Support touch linking;
- Show `config.group` in the REST API (although some switches will send broadcasts after reset, and will need to be bound to a group manually befor showing `config.group`);
- Show client (grey) _On/Off_, _Level_, and/or _Scenes_ clusters in the deCONZ GUI (although with Xiaomi there doesn’t seem to be any relationship between the clusters they advertise vs the clusters they actually support).

rendgeor, 2019-09-02:
Hello,
I bought the switch ćause thought they are fully supported by deconz.
Adding the switch results in no new switch object in phoscon.

Anyhow, I thing it was added by isn't shown.

Is that the icasa switch and if so, why isn it shown in phoscon after parting?
[image]

Btw:
Do pairing with:
1) phoscon: search for switches
2) icasa switch: on + off, then on...blinks

rendgeor, 2019-09-04:
@manup 
Switch is seen in webapp. Why isn it available in phoscon?

[image]

closed 2020-01-09 by stale[bot]

alexdepalex, 2020-02-14:
Same here als @rendgeor. Switch shows up on Phoscon. I also see events being registered in the Deconz log, but the switch doesn't show up in the web app and HA. Any idea what's needed to achieve this?

manup, 2020-02-14:
The iCasa (Sunricher) switches will be added in the Phoscon App and Switch Editor in the coming releases. Note that the old WebApp is no longer actively maintained. For the HA part I don't know what is missing.

alexdepalex, 2020-02-14:
Thanks @manup. What's the eta on that release?

manup, 2020-02-14:
No ETA yet, but we're figuring out to provide a beta URL for the Phoscon App in the coming weeks, which provides more device integrations than the current stable/official version.

huubke80, 2020-02-22:
Nice I'm looking forward to the new release. Will only the switches be added in the new release, or also the LED drivers from (Sunricher).

alexdepalex, 2020-02-22:
@huubke80 If you’re using HA, the latest release of the deconz add on supports the switches.