dresden-elektronik/deconz-rest-plugin #1128: deconz crashing when adding motion sensor

deheingit, 2019-01-13:
Hi,
I have a issue with deconz. Today I wanted to add a new Xiaomi Awara Motion sensor (https://de.gearbest.com/alarm-systems/pp_659226.html). But every time i try to add the sensor the web ui is not reachable anymore and I have to restart deconz. It works with trying it a couple if times. Once connected the sensor works fine.

I’ve uploaded the log here: https://pastebin.com/WB5YZUBF

manup, 2019-01-13:
If possible can you please create a core dump while deCONZ crashes?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Enable-core-dumps-on-Raspbian

manup, 2019-01-16:
This should be fixed with 2.05.55

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_55

Closing the issue for now.

closed 2019-01-16 by manup