dresden-elektronik/deconz-rest-plugin #1131: Xiaomi RHT sensors stop reporting

djwlindenaar, 2019-01-14:
I'm on 2.5.54 and as of a couple of days sometimes the RHT sensors of xiaomi fail to report. This is new, as they worked fine up to now. I have both with and without pressure sensor (round and square). They both show the behaviour. 

They are not truly lost, when you press the button, the report comes through correctly...

Anything I could test/try?

Best regards
Daniel

manup, 2019-01-14:
Some questions:
* Do you have the deCONZ gui running?
* To which parent node are the sensor connected?
* Which firmware version are you running?

Sometimes it helps to press the small button on the sensor 5 times quickly to start a signal test on the device.

djwlindenaar, 2019-01-15:
- Yes
- not clear, no line is drawn (also not for sensors still working)
- 262F0500

I'll try that trick, what effect is expected from the signal test?

Best regards
Daniel

djwlindenaar, 2019-01-16:
It did not help... It looks like there are reports that get through, but only the ones sent upon >0.5C temperature change. But not the hourly updates these devices also send. 
Does that make sense or could it just be random ones getting through and me misinterpreting

ebaauw, 2019-01-16:
The report sent ad hoc, on larger temperature changes are standard _Temperature Measurement_ cluster attribute reports, whereas the hourly reports are Xiaomi-specific _Basic_ cluster attributes reports.  The hourly reports differ per model, so it could very well be that deCONZ doesn't understand some of them.

When starting deCONZ with `--dbg_aps=2`, it logs the incoming attribute reports.
For the change reports looks something like:
```
Jan 16 06:52:32 pi2 deCONZ[12654]: 06:52:30:897 APS-DATA.indication srcAddr: 0x00158d0001a4b6ed, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0402, lqi: 255, rssi: -59
Jan 16 06:52:32 pi2 deCONZ[12654]: 06:52:30:897     asdu: 180b0a000029bb07
Jan 16 06:52:32 pi2 deCONZ[12654]: 06:52:30:899 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
Jan 16 06:52:32 pi2 deCONZ[12654]: 06:52:30:899 ZCL attribute report 0x00158D0001A4B6ED for cluster 0x0402, ep 0x01
```
For the hourly (more or less) reports, it looks something like:
```
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:219 APS-DATA.indication srcAddr: 0x00158d0001a4b6ed, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -58
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:219     asdu: 18070a01ff422501213b0b0421a84305215c010624010000000064297d076521bb12662b148901000a210000
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:220 no button map for: lumi.weather ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 001
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:220 ZCL attribute report 0x00158D0001A4B6ED for cluster 0x0000, ep 0x01 
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:220 0x00158D0001A4B6ED extract Xiaomi special attribute 0xFF01
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:220     01 battery 2875 (0x0B3B)
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:221     04 unknown 17320 (0x43A8)
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:221     05 RSSI dB (?) 348 (0x015C)
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:221     06 LQI (?) 4294967296 (0x0100000000)
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:221     64 temperature 1917
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:221     65 humidity 4795
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:221     66 pressure 1006
Jan 16 06:42:32 pi2 deCONZ[12654]: 06:42:30:221     0a unknown 0 (0x0000)
```

If you can capture this info for your sensors (case insensitive search for the sensor's mac address in the log) we can check whether deCONZ understands the special report.

ebaauw, 2019-01-16:
> I've been using them for over a year now

Did you check the battery?

manup, 2019-01-16:
That's strange, mine also stopped working. I didn't noticed it. The last timestamp is a week ago.

For testing I've pressed the small button once, the sensor looks for a network and didn't try to rejoin as it it was factory new state.

So I've startet sensor search and pressed the small button again once. The sensor joined immediately and was reachable again. It does have a new nwk address. 

I wonder what brought the sensor to factory settings.

djwlindenaar, 2019-01-17:
I had one in the same state. The outside temp sensor. After doing a factory reset and rejoin, it happily reported. Especially during the ~10 minutes it takes to cool down again to the outside temperature, but after that silence. 
I'll replace the battery just to be sure, but seeing that the attribute reports are sent, but not the special reports seems strange to attribute to low battery

ebaauw, 2019-01-17:
Double-checked my three `lumi.weather` sensors.  Indeed, the bathroom sensor hasn't been sending hourly reports for at least three weeks.  I hadn't noticed before, since `state.lastupdated` did change for the ad-hoc reports, but the Eve history shows infrequent changes.  Reset and re-paired the sensor yesterday evening, and I've been receiving reports since then.  Battery reports 2.895V, which is higher than one of the other sensors that has been sending the special reports.

djwlindenaar, 2019-01-21:
During the weekend I reset and rejoined all silent sensors and they are now again doing hourly reports. 
I noticed that the outside sensor was a different situation altogether as it was filled with water. After drying, it's still working and now also reporting happily.

Still the question remains what caused all of these sensors to stop hourly reporting?

manup, 2019-01-21:
>Still the question remains what caused all of these sensors to stop hourly reporting?

Which firmware version is shown for the sensors?

For the Mi version there were some fixes to improve the hourly reports in 2.05.55 and upwards.

djwlindenaar, 2019-01-21:
Some say 20160516 (round ones), some 20161129 (square ones)

I'm running .55 as of Saturday, so we'll see how they hold up.

manup, 2019-01-21:
Cool, please keep us informed if it works or not.

djwlindenaar, 2019-01-21:
[image]

I guess my outside temp sensor is gone after all. It's cold, but not that cold...

manup, 2019-01-21:
:) haha, cool (indeed).

What kind of sensor is it?

djwlindenaar, 2019-01-22:
That's the square one. lumi.weather

djwlindenaar, 2019-01-27:
Looks like the issue is indeed fixed in .55 and .57

closed 2019-01-27 by djwlindenaar