dresden-elektronik/deconz-rest-plugin #1137: Binding between switch and lights to control them when DeCONZ is offline, when and how?

JBS5, 2019-01-15:
During a investigation which parts of my smart home will work when HASS and DeCONZ are offline, I am wondering when and how a binding between a switch (e.g. Hue Dimmer Switch) and one or more (e.g.) IKEA Tradfri bulbs is made.

I assume there will be a binding between them when adding one or more bulbs to a Hue Dimmer Switch group in the Phoscon app? And because of this binding, I am able to control those bulbs with this Hue Dimmer Switch when DeCONZ and HASS are offline, is this correct?

After adding some other Tradfri bulbs to another (Hue Dimmer Switch) group, I am not able to control those bulbs with that Hue Dimmer Switch when HASS and DeCONZ are offline. I assumed this should work (like the other Dimmer Switch group which I described above), and I am wondering why not. Does it take some time to make the bindings? Can I force this? Or, don't I understand the whole binding part?

I am using this docker https://github.com/marthoc/docker-deconz on a NUC with HASS and a Conbee.

jo-me, 2019-01-16:
+1 

I have a Hue Dimmer Switch and a Hue bulb that work together when deconz is down but I don't know what I did to achieve this..
The originally came bundled as kit (without bridge) and I paired them to deconz afterwards.

ebaauw, 2019-01-16:
The switch doesn’t bind to any of the lights; it only binds to the group.  That is, the dimmer is configured to send group commands (for on/off, brightness, ...) on button actions.   A group in Zigbee is very much like a multicast address.  The bulbs the “subscribe” to the group.

Once setup, this works without interference from the gateway, also when the gateway is down.  In my experience, it could take some time before the gateway tells the light to join the group.  You can check this in the deCONZ GUI, in the _Cluster info_ panel for the _Groups_ cluster (of the light).

ebaauw, 2019-01-16:
Yes.  Some switches (typically those that support touch linking) send commands to bound groups and/or lights; others, like Xioami send reports to the coordinator (very much like sensors).  The Hue dimmer actually does both.

JBS5, 2019-01-16:
Thanks again!

Last thing I am wondering (for now ;-)): When using automations in HASS with deCONZ events from a Hue Dimmer Switch as triggers and turnOn/turnOff as actions, What does control the lights when deCONZ and HASS are online? The binding, or the HASS automation?

ebaauw, 2019-01-16:
Both actually.  The group command probably arrives first, so the rule command sticks.

ebaauw, 2019-01-16:
The light will reflect to the rule command, as this overrides the changes from the group command.

JBS5, 2019-01-18:
> Once setup, this works without interference from the gateway, also when the gateway is down. In my experience, it could take some time before the gateway tells the light to join the group. You can check this in the deCONZ GUI, in the _Cluster info_ panel for the _Groups_ cluster (of the light).

I am not sure how to check this. When opening the _Cluster info_ panel when a light (which can be controlled when deCONZ is offline) is selected, I see the info below which didn't looks like a proper group? 

[image]

ebaauw, 2019-01-18:
If you execute the _Get group membership_ command, it returns the _Capacity_, _Group Count_, and _Group List_.  Unfortunately the list only displays the first group, but the count increases and the capacity decreases with each new group.  You can also use the _View Group_ command, to check whether the light subscribes to a particular group.  Enter the group ID (from the REST API), converted to hex and check for _Status_ `SUCCESS`.

closed 2019-05-25 by stale[bot]

thedannymullen, 2019-08-25:
When I click on cluster info it is blank for all devices in deconz GUI.  Is there a trick to making it show?

Smanar, 2019-08-26:
blank ?
You have screenshoot ?

Smanar, 2019-08-27:
It's normal.
The 5 button can make "group request" to works even the gateway is off, so this device have a group just for him, all the lights you put on this group, will works even deconz is off.
The Xiaomi don't have this feature, so you need to make a group ( a normal one) yourself.

ebaauw, 2019-08-28:
Xiaomi switches simply don’t send commands to control lights directly.  They only work with a gateway or hub.  No amount of grouping can change that.

ebaauw, 2019-08-28:
Sorry, my bad.  What switch do you have?  Does it show `config.group` in the REST API?

thedannymullen, 2019-08-29:
@ebaauw I am using the ikea on off switch the square one. It comes with the tradfri plug.

ebaauw, 2019-08-29:
The REST API plugin creates a group for that switch on pairing, all you need to do is assign lights to that group.  You might need to upgrade deCONZ and re-pair the switch.

thedannymullen, 2019-08-29:
Ok here is where I am at..   I do a get on group 8 in the rest api.  It returns this: 
`{"action":{"bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":true,"sat":127,"scene":null,"xy":[0,0]},"devicemembership":["3"],"etag":"ef88cd7f164e537e4e770f08f5343b5b","id":"8","lights":[],"name":"TRADFRI on/off switch ","scenes":[],"state":{"all_on":false,"any_on":false},"type":"LightGroup","uniqueid":"d0:cf:5e:ff:fe:72:03:92"}`

I see that there are not lights defined in this group. I am making the assumption that I need to add the lights into the (lights:[]) area.  I have tried to do that using this post command:

`{"application":"my` application","action":{"bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":true,"sat":127,"scene":null,"xy":[0,0]},"devicemembership":["3"],"etag":"ef88cd7f164e537e4e770f08f5343b5b","id":"8","lights":["1"],"name":"TRADFRI on/off switch `","scenes":[],"state":{"all_on":false,"any_on":false},"type":"LightGroup","uniqueid":"d0:cf:5e:ff:fe:72:03:92"}`

I received a success back from the API.   However when I query the group again I dont see the light added.  

I am using Postman app on windows to issue the post commands.  Curious If you can clue me in to the syntax I need to post to the rest API.  I have used it to create other things successfully in Postman, but can't seem to figure out how to modify only one parameter.  

Thanks!

thedannymullen, 2019-08-30:
**UPDATE I got it!!!**

I used the wrong command above.  To make it work I used a put command vs post.  This is my first time interacting with rest api.   I used {"application":"my application","lights":[ "1" ]}"

@ebaauw @Smanar thanks for help!!