dresden-elektronik/deconz-rest-plugin #1138: IKEA Trådfri, same bulbs but different default temperatures when powered-on

mwhrtin, 2019-01-16:
Running deCONZ 2.05.55 with a Conbee stick, latest dev package and Rest API on Raspbian. I've got three IKEA Trådfri color bulbs, Phoscon reports them as *"TRÅDFRI bulb E27 CWS opal 600lm"*. 

Two of the bulbs report firmware 1.3.002 in deCONZ and one of them firmware 13000272. Not sure if it's the same version, but none of the bulbs want to update to 1.3.000272 (latest I can download) when I try to manually trigger OTAU.

All three bulbs will power-on to the last used brightness, but one of the bulbs (reports version 1.3.002) will default to a semi-warm temperature (around 2500-2700k), but two of the others (both 1.3.002 and 13000272) defaults to a cold temperature (around 4000k). 

Anyone know why is the color temperature used when powered-on is different for the same bulbs, and is this something that can be changed/controlled in deCONZ? I don't find any differences between the bulbs in the cluster info panel or anywhere else.

rikroe, 2019-01-26:
Are you using them in a group? I had the issue that when I tried to set brightness and temperature via the group control, sometimes it does not get applied (especially when using scenes). 

So I had to change to toggle from group to single light and adjust each light manually. Afterwards I was also able to change back to group.

mwhrtin, 2019-01-26:
I'm not using them in a group. Yesterday I reset all my light bulbs (for another reason) and also re-synced them to deCONZ. 

After they were reset they all got the warmer 2700K color, and since then they've used it as the default when powered-on. 

Closing this issue.

closed 2019-01-26 by mwhrtin