dresden-elektronik/deconz-rest-plugin #1140: v2.05.55: no ZigBee traffic

ebaauw, 2019-01-16:
Upgraded my test network to v2.05.55 yesterday.  Still on 0x262f0500.  I just saw that all routers on my test network (IKEA colour bulb, iCasa dimmer, two XBees) show red in the GUI.  Tried all the usual tricks to revive them: Press `0`, Power cycled the routers, restarted deCONZ, reset the RaspBee, power cycled the Raspberry, started deCONZ with all routers powered down, and added them one by one: no joy.  The iCasa keypads still control the IKEA bulb.

The network settings seem valid.  Checking the log, deCONZ only receives data from the RaspBee.  Only requests to the RaspBee get confirmed; requests to other nodes get confirm messages with status 0xA7 NO_ACK or status 0x0D.  Sniffing, the RaspBee doesn't seem to send any ZigBee traffic.

My best guess: a hardware/NVRAM issue in the RaspBee?  Please advise.

manup, 2019-01-16:
Hmmm not good, but fixable manually. Can you please check the zll.db file table `zbconf` Here you can see the network settings which are recorded if they change. so you can see if there is a difference in 2.05.52/2.05.55.

You can then enter the settings from 2.05.52 manually in deCONZ Network Settings, afterward
* Save
* Done
* Leave
* Join

In the past weeks I've studied many database files where the parameters had changed (reason is unknown) often network key/panid or channel were corrupted somehow. I hope to put out some auto repair to fix settings if they were changed without user intention. Or at least have a page where settings can be reviewed and easily repaired.

ebaauw, 2019-01-16:
Upon double checking, it would seem that the network key is corrupted.  Suspiciously many zeroes, and this key is not in my WireShark config.

[image]

manup, 2019-01-16:
Possible, if the network key is different in older table entries than this is likely the cause.

closed 2019-01-16 by ebaauw

manup, 2019-01-16:
> This is pretty scary...

It is, I'm not sure where this happens (likely firmware) the fix for this will be done in deCONZ first since it has the broader picture about which are the correct settings.

ebaauw, 2019-01-16:
Thanks, as always!