dresden-elektronik/deconz-rest-plugin #1141: No WebUI or Zigbee devices in 2.05.55

spridarn, 2019-01-16:
I updated my hass.io Pie to 2.05.55 earlier today. After the upgrade all my Zigbee devices are missing in Home Assistant. The WebUI for deCONZ is dead. If I check the log under the add-on I see that several devices talk but nothing more. Before I upgraded I made a snapshot in hass.io, I have tried to roll back to it but still the same problem. If I check the system log it still says this:
19-01-16 21:55:20 INFO (SyncWorker_6) [hassio.docker.addon] Start Docker add-on marthoc/hassio-addon-deconz-armhf with version 2.05.55
Even if I run a snapshot before 2.05.55. 
Saw that when rebooting the old addon now the addon log says this:
[Hass.io] Starting in minimal mode, without VNC GUI

And that was introduced in 2.05.55. Anyone know how to totally remove all files from the plugin so my snapshot maybe work?

Webserve, 2019-01-16:
https://github.com/marthoc/hassio-addons/issues/64

schisamo, 2019-01-16:
This isn't a deCONZ issue but instead is a bug in the Home Assistant deCONZ addon. The underlying issue is fixed in https://github.com/marthoc/docker-deconz/pull/93

spridarn, 2019-01-16:
> This isn't a deCONZ issue but instead is a bug in the Home Assistant deCONZ addon. The underlying issue is fixed in [marthoc/docker-deconz#93](https://github.com/marthoc/docker-deconz/pull/93)

Ok thanks for info, waiting for it to be populated to the hass.io addon store then

corvy, 2019-01-17:
Quick fix is to delete the integration and set it up again. Phoscon is on port 80.

chimpy, 2019-01-17:
This should be closed.

closed 2019-01-17 by manup