dresden-elektronik/deconz-rest-plugin #1144: Aqara temperature/humidity sensor

erikproper, 2019-01-17:
Dear all,
  I've purchased two Aqara/Xiaomi temperature/humidity sensors (https://www.gearbest.com/goods/pp_009571080730.html?lang=en&wid=1433363&utm_source=email_sys&utm_medium=email&utm_campaign=GB_GB_orderShippedOut_190105_AI_750_254).

They are the (newer?) "square" ones. I can't seem to connect them to deconz. 
The Supported Devices does list:
- Xiaomi | Mi temperature/humidity sensor | Tb, Hb
But I guess this is the (older?) round one.

In the GUI app, I do see two new devices, with Temperature (0402), Humidity (0405) and Pressure (0403) attributes:
[image]

I'm running version 2.05.52 of Deconz.

Cheers,
  Erik

manup, 2019-01-17:
>They are the (newer?) "square" ones. I can't seem to connect them to deconz.
The Supported Devices does list:

Both, the Mija version (round) and Aqara version (square) are supported.

[image]

Please rejoin the sensors with latest deCONZ version 2.05.55, there were quite some fixes related to joining.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_55

Note that you need to start sensor search in the Phoscon App to let the sensors join and related resources will be created.

erikproper, 2019-01-17:
Trying this as we speak. No luck yet ... will keep on trying (like with the
door sensors ...)

As a side issue ... when listing the sensors via http://
<hub>/api/<key>/sensors,
each sensor has a number (1, 2, 3, etc), which allows you to get more info
via sensors/<nr>.
That number ... does that remain stable? I have a suspicion that in between
updates of the
deconz software, this number changes. Could that be the case?

Software, like FHEM (www.fhem.de) uses that number (i.e. not the UniqueID)
to identify the HUE device
from within FHEM.

Cheers,
  Erik

erikproper, 2019-01-17:
Something is wrong with my Deconz setup it seems.
1] Missing several of the other sensors when looking at the sensor page in
the Phoscon app.
The sensors still show up in the REST API though

2] Tried removing an existing sensor (Phillips motion + lux + temp), and
re-add it. No luck.
It blinks red regularly. Tried resetting it. Does not join anymore.

chimpy, 2019-01-17:
> Something is wrong with my Deconz setup it seems. 1] Missing several of the other sensors when looking at the sensor page in the Phoscon app. The sensors still show up in the REST API though

I don't suppose you're running deconz and deconz-gui? Take a look through this thread and try analysing your system with the commands therein: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1117

erikproper, 2019-01-17:
Okay. Found some issues, but still stuck at the original situation.

As it turns out, there was a magnet sensor registered as a switch. I removed that one manually via the REST interface. That brought all the original sensors back.
After a while, I was able to re-register the Philips Motion sensor. 

Now back to the new sensors ...

erikproper, 2019-01-17:
It now worked ....

closed 2019-01-17 by erikproper

kugelkopf123, 2019-01-18:
No. Not Really not as it does earlier. It is like they dont react as good as it was with FW <.50. I do not know if possibly the brightness has an influence on it?! 

I have added them several times via Phoscon. Stays the same... It would really nice if you could choose the time by yourself. Perhaps that would help... 
90sec. is really too long for me. And why does the presence value do not stay "true" for the whole 90 sec?