dresden-elektronik/deconz-rest-plugin #1145: Xiaomi motion sensors wrong state

rtenklooster, 2019-01-17:
Hi, 
Since I’ve updated to 2.05.54 my motion sensors report motion while there is no motion.  Rest api also reports the wrong state. 

Anyone else experiencing these issues? Have rebooted, powered off, nothing helps. 

I have made a backup before upgrading, does restoring a backup reverts the version?

AndreasRavn, 2019-01-17:
Known bug in 2.05.54, upgrade to 2.05.55 and its fixed.

rtenklooster, 2019-01-17:
Thanx, I’m wondering why the update isn’t automatically available. I’m using the Deconz image, it’s no problem there is a bug, however I would expect the update would be published like the .55 update was.

manup, 2019-01-17:
>Thanx, I’m wondering why the update isn’t automatically available. I’m using the Deconz image, it’s no problem there is a bug, however I would expect the update would be published like the .55 update was.

The updates via the update button are published delayed when the version is considered stable and available for a few days via manual install.

closed 2019-05-24 by stale[bot]