dresden-elektronik/deconz-rest-plugin #1147: deconz.service fails!

Calimeistro, 2019-01-18:
So I've installed Deconz on Ubuntu linux, and trying to get autostart headless to work.
Command: sudo systemctl enable deconz

Checking status by: sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/etc/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2019-01-18 01:52:38 CET; 1min 30s ago
  Process: 1029 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code=exited, status=217/USER)
 Main PID: 1029 (code=exited, status=217/USER)

jan. 18 01:52:38 cali-HP-ProBook-645-G1 systemd[1]: deconz.service: Main process exited, code=exited, status=217/USER
jan. 18 01:52:38 cali-HP-ProBook-645-G1 systemd[1]: deconz.service: Failed with result 'exit-code'.
jan. 18 01:52:38 cali-HP-ProBook-645-G1 systemd[1]: deconz.service: Service hold-off time over, scheduling restart.
jan. 18 01:52:38 cali-HP-ProBook-645-G1 systemd[1]: deconz.service: Scheduled restart job, restart counter is at 5.
jan. 18 01:52:38 cali-HP-ProBook-645-G1 systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
jan. 18 01:52:38 cali-HP-ProBook-645-G1 systemd[1]: deconz.service: Start request repeated too quickly.
jan. 18 01:52:38 cali-HP-ProBook-645-G1 systemd[1]: deconz.service: Failed with result 'exit-code'.
jan. 18 01:52:38 cali-HP-ProBook-645-G1 systemd[1]: Failed to start deCONZ: ZigBee gateway -- REST API.

manup, 2019-01-18:
Which deCONZ version is this? 

Currently the service expects that the userid 1000 exists (first user created on a Linux machine).
Can you please check in a terminal that the user with 1000 exists.

    getent passwd 1000

Note to self: provide overwrite wiki or automatic resolve.

Calimeistro, 2019-01-18:
Hey,
getent passwd 1000 provides me with: cali:x:1000:1000:Terje Hansen,,,:/home/cali:/bin/bash

My service file:
[Unit]
Description=deCONZ: ZigBee gateway -- REST API
Wants=deconz-init.service deconz-update.service

[Service]
User=pi
ExecStart=/cali/bin/deCONZ -platform minimal --http-port=80
Restart=on-failure
StartLimitInterval=60
AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME

[Install]
WantedBy=multi-user.target

I belive this is version 2.05.20, that was the one I found in my download folder and 99% sure that I installed.

manup, 2019-01-18:
>I belive this is version 2.05.20, that was the one I found in my download folder and 99% sure that I installed.

The issue here seems to be that the old version uses the Raspberry Pi user pi which doesn't exist on Ubuntu. Please try with version 2.05.55, the issue was fixed a while ago. 

https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages

Calimeistro, 2019-01-19:
> The issue here seems to be that the old version uses the Raspberry Pi user pi which doesn't exist on Ubuntu. Please try with version 2.05.55, the issue was fixed a while ago. [...]

Done 👍 
However the file remained the same, this is the config;

[Unit]
Description=deCONZ: ZigBee gateway -- REST API
Wants=deconz-init.service deconz-update.service

[Service]
User=pi
ExecStart=/cali/bin/deCONZ -platform minimal --http-port=80
Restart=on-failure
StartLimitInterval=60
AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME

[Install]
WantedBy=multi-user.target

closed 2019-01-19 by Calimeistro