dresden-elektronik/deconz-rest-plugin #1148: Updating color through Rest API does not trigger colormode event (IKEA Tradfri)

mwhrtin, 2019-01-18:
Running deCONZ 2.05.55 and Rest API 1.16.0.

**Issue:** 
Changing color using the Rest API does not send change events for `state.colormode` if the  value is changed.

**Description:** 
As stated in other reported issues, the IKEA Tradfri RGB light bulbs changes the color mode to the unsupported `ct` when they are powered-on. 

When the light is in `colormode: ct` updating the color through the Rest API will change both `state.xy` and `state.colormode` but only the `state.xy` events are sent over the websocket. When updating the color using a rule instead it sends change events for both `state.xy` and `state.colormode`.

This affects how the light is registered in HomeAssistant, when the light is in `colormode: ct` no color attributes are displayed. Updating the color through the Rest API will change the color mode to `xy`, but this is never sent back to HomeAssistant. HomeAssistant still believes the light is in the invalid `state.colormode: ct`.

**Example:**
Rule that will send both `state.colormode` and `state.xy` events over the websocket if the light's `state.colormode` changes from `ct` to `xy` when the color is changed.

```json
{
  "actions": [
    {
      "address": "/lights/3/state",
      "body": {
        "xy": [ 0.2, 0.2 ]
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/lights/3/state/on",
      "operator": "eq",
      "value": "true"
    },
    {
      "address": "/lights/3/state/on",
      "operator": "dx"
    }
  ]
```

Events received when the rule is triggered:

```json
{"e":"changed","id":"3","r":"lights","state":{"colormode":"xy"},"t":"event","uniqueid":"90:fd:9f:ff:fe:f2:f6:19-01"}
{"e":"changed","id":"3","r":"lights","state":{"x":13055},"t":"event","uniqueid":"90:fd:9f:ff:fe:f2:f6:19-01"}
{"e":"changed","id":"3","r":"lights","state":{"y":13055},"t":"event","uniqueid":"90:fd:9f:ff:fe:f2:f6:19-01"}
```

Rest API request that will only send state.xy events over the websocket, despite `state.colormode` changing from `ct` to `xy`:

**PUT /lights/3/state/**
```json
{
  "xy": [ 0.2, 0.2 ],
  "on": true
}
```

Events received when the Rest API request is sent:

```json
{"e":"changed","id":"3","r":"lights","state":{"x":13055},"t":"event","uniqueid":"90:fd:9f:ff:fe:f2:f6:19-01"}
{"e":"changed","id":"3","r":"lights","state":{"y":13055},"t":"event","uniqueid":"90:fd:9f:ff:fe:f2:f6:19-01"}
```

manup, 2019-01-18:
Good catch, the color handling is currently in refactoring. I guess we should do two things here:
* Provide the `state.colormode` changes due REST-API actions as event
* Always set `state.colormode` to `"xy"` for the Ikea CWS bulbs

The `ct` issue should be tracked over time when newer firmware becomes available.

manup, 2019-01-20:
Fixed in 2.05.57, closing the issue for now.

closed 2019-01-20 by manup