dresden-elektronik/deconz-rest-plugin #1150: deConz do not find hue bulbs

Monnem, 2019-01-18:
Hi all,
have here a RP3+ with usb-Stick.
Version 2.05.54 / 4.1.2019
Firmware 262F0500 
(Same issue also with 2.5.50)

Got some new hue WW/CW buls in two starter kits (with bridge and dimmer)
Reseted with dimmer several times every bulb (with blinking confirmation ) and tried to bind them to deConz. Only on bulb out of four was found as:
ModellHue A19 White Ambiance
Version 1.29.0_r21169
I extended the search time to 10 min with no success
The one bulb was found immediately. The other ones still fail. Reseted them several times, open network multiple times ans extended also search time to 10 min.
Right now I'm  "a little bit" frustrated. Any hint is welcome! ;-)

Thanks in advance
Monnem

Monnem, 2019-01-18:
add: the bulbs are in pairs in a lamp. One lamp is 1m theother 2m away. Linked bulb in the first one.

ebaauw, 2019-01-18:
Some hints:
- Make sure to power cycle the bulb after resetting it.  With recent firmware, they're only discoverable for some time after power-on;
- Use a USB extension cable to put some distance between the Raspberry and the ConBee.  There's usually a lot of interference when the ConBee is plugged directly into a computer, especially with WiFi and/or Bluetooth enabled;
- What type of lamps do you have?  Metal enclosures can block the Zigbee radio signals;
- Move the bulbs and the ConBee closer to one another.  If the light enclosures are fixed, move the ConBee instead, by using a longer USB extension cable (mine is 5m), or by powering the RaspBee from a USB power bank.

Monnem, 2019-01-21:
Hi,
thanks for your hints. 
Finally a usb extension cable to reduce the distance down to 10cm was the solution.
Best
Monnem

closed 2019-01-21 by Monnem