dresden-elektronik/deconz-rest-plugin #1151: Xiaomi Aqara Magic Cube 'buttonevent' is exposed wrong via REST interface

cweitkamp, 2019-01-18:
I connected a Xiaomi Aqara Magic Cube to my deCONZ via RaspBee (Version 2.05.55 / 14.1.2019
Firmware 262F0500). The `buttonevent` property on the REST interface exposes the wrong value (rotation in degrees?) whereas it is shown correctly in Phoscon App.

Start state:

[image]

```
{
    "config": {
        "battery": 75,
        "on": true,
        "reachable": true,
        "temperature": 1900
    },
    "ep": 3,
    "etag": "bf80937fb2bd06257aaa9f6b8635c864",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.sensor_cube.aqgl01",
    "name": "Zauberwürfel",
    "state": {
        "buttonevent": 334,
        "lastupdated": "2019-01-18T20:47:15"
    },
    "swversion": "20160704",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:XX:XX:XX-03-000c"
}
```
If I flip and turn the cube the state changes to:

[image]

```
{
    "config": {
        "battery": 75,
        "on": true,
        "reachable": true,
        "temperature": 1900
    },
    "ep": 3,
    "etag": "731c6267efcabdbb10e45a724160946a",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.sensor_cube.aqgl01",
    "name": "Zauberwürfel",
    "state": {
        "buttonevent": 3843,
        "lastupdated": "2019-01-18T20:50:39"
    },
    "swversion": "20160704",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:XX:XX:XX-03-000c"
}
```

ebaauw, 2019-01-18:
The two ZHASwitch resources for the cube: one with regular button events for the digital events (flip, push, shake, tap); and one for the analog events (turn).  You're looking at the analogue one; Phoscon is looking at the digital one.

cweitkamp, 2019-01-19:
Yes, that is true. I totally missed that. Thanks for reminding me.

Should we maybe add a small remark about that in the [supported devices section](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#supported-wireless-switches)?

closed 2019-01-19 by cweitkamp