dresden-elektronik/deconz-rest-plugin #1153: Make wifi on frontend visually configurable

Kane610, 2019-01-19:
For containers there is no reason to show wifi in the frontend. Could you make this configurable as to not be visible in setups where it is not relevant

manup, 2019-01-20:
As first step in 2.05.57 the deconz-wifi service is not loaded and displayed when used with Docker containers. In next versions we'll add a easier way to disable the service via Phoscon App or REST-API.

Kane610, 2019-01-21:
Solved with latest release!

closed 2019-01-21 by Kane610

Ronoon, 2019-02-18:
@manup 
Looks like you´ve done a bit to much this time cause without changing my WiFi settings they disappeared completly on my Raspi 3+ on version .59 maybe before but i didn noticed.
Before this change i could switch between Wifi and Host mode ... right now i have a blank page as where my settings belonged to
Maybe you could have a look at that.