dresden-elektronik/deconz-rest-plugin #1154: Offset problems

cosmo84, 2019-01-20:
Referring to #763 I have set an offset of -2 degrees for one of my Xiaomi temperature sensors. The offset itself is working, but it is not respected every few updates, see the following logs:

[1/20/2019, 6:32:26 AM] [Phillips Hue] Draußen: set homekit temperature from -0.4°C to -2.3°C
[1/20/2019, 6:42:32 AM] [Phillips Hue] Draußen: set homekit temperature from -2.3°C to -0.4°C
[1/20/2019, 7:13:28 AM] [Phillips Hue] Draußen: set homekit temperature from -0.4°C to -2.5°C
[1/20/2019, 7:38:32 AM] [Phillips Hue] Draußen: set homekit temperature from -2.5°C to -0.5°C

So it seems only every second measuring is with the offset

ebaauw, 2019-01-20:
Looking at the code, `config.offset` is ignored when updating `state.temperature` (or `state.humididy`) from the Xiaomi special report.  Note that it should not be applied when updating `config.temperature`, but likely `config.offset` wouldn’t be present anyways for those resources.

cosmo84, 2019-01-20:
Sorry, I don‘t get the point yet. Am I using the offset in a wrong way, or is it not intended to work like this at all?

ebaauw, 2019-01-20:
No, sorry, there's an omission in de REST API plugin.  Preparing a fix...

cosmo84, 2019-01-20:
All right, thanks a lot.

manup, 2019-01-20:
Please checkout if it works now with 2.05.57, closing the issue for now.

closed 2019-01-20 by manup

kugelkopf123, 2019-03-23:
Hey. 

Sorry that I have to reopen this Issue. Its just a Question. Where and how can I set this offset. Because I have such a Aqara TempHumidityPressure Sensor, too. It would be nice to set an offset on the temperature. But i can not find it in the Deconz GUI nor Phoscon. 
[image]
[image]

Thanks for your help!!