dresden-elektronik/deconz-rest-plugin #1157: ConBee disconnecting every 15 seconds.

dape82, 2019-01-20:
Hello,
when I first tried deconz I had the same problem. I was asked to install the raspberry image. I tried it and it worked fine until a few days (or weeks?) ago. I do not know when it started excactly.

In the GUI I can see that deconz disconnects every 15 seconds. 
Is this a known problem? Is there any fix? Are there logs somewhere where I can check why it keeps disconnecting. 

During 10 seconds between disconnects  everything seems to work fine.
I have a Hue remote, a few Xiaomi temperature sensors and a Xiaomi motion sensor attachted via the Phoscon App.

I tried updating deconz to 2.05.55. Same issue.
The deconz firmware version is 262F0500.

Thanks in advance!

Regards,
Daniel

manup, 2019-01-20:
Hmm can you please check if not multiple instances are running:

    ps ax | grep deCONZ

Might also be worth to check if there are further services running which may access the USB serial port.

dape82, 2019-01-20:
This is the output from the startup to the first reconnect.

```
19:59:27:787 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
19:59:27:787 CTRL. 3.16.219:59:27:928 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
19:59:28:195 parent process bash
19:59:28:195 gw run mode: normal
19:59:28:196 GW sd-card image version: Phoscon-GW, V4_00, 2018-10-30
19:59:28:196 sd-card cid: 1b534d454434515430ffffffff012b49
19:59:28:198 DB sqlite version 3.16.2
19:59:28:199 DB PRAGMA page_count: 30
19:59:28:199 DB PRAGMA page_size: 4096
19:59:28:200 DB PRAGMA freelist_count: 0
19:59:28:200 DB file size 122880 bytes, free pages 0
19:59:28:200 DB PRAGMA user_version: 6
19:59:28:200 DB cleanup
19:59:28:200 DB create temporary views
19:59:28:228 don't close database yet, keep open for 900 seconds
19:59:28:229 started websocket server at port 443
19:59:28:234 discovery updated announce interval to 10 minutes
19:59:28:238 found node plugin: libde_rest_plugin.so - REST API Plugin
19:59:28:243 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
19:59:28:265 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
19:59:28:555 discard sensor state push for 1: state/dark (already pushed)
19:59:28:557 discard sensor state push for 1: state/status (already pushed)
19:59:28:558 discard sensor state push for 1: state/lastupdated (already pushed)
19:59:28:930 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:28:930 device state timeout ignored in state 2
19:59:29:182 UPNP socket not bound, state: 0
19:59:29:466 could not delete /var/tmp/deconz-firmware-version
19:59:29:584 Device firmware version 0x262F0500
19:59:29:610 unlocked max nodes: 200
19:59:29:898 Device protocol version: 0x0109
19:59:29:934 new node - ext: 0x00212effff022cc7, nwk: 0x0000
19:59:30:042 SensorNode 4 set node 0x00158d0002375fe6
19:59:30:042 SensorNode 2 set node 0x00158d0002375fe6
19:59:30:043 SensorNode 3 set node 0x00158d0002375fe6
19:59:30:086 SensorNode 5 set node 0x0017880104e6e470
19:59:30:104 SensorNode 6 set node 0x00158d0002b5c578
19:59:30:105 SensorNode 8 set node 0x00158d0002b5c578
19:59:30:105 SensorNode 9 set node 0x00158d0002b5c578
19:59:30:126 SensorNode 10 set node 0x00158d0002b7eea3
19:59:30:127 SensorNode 11 set node 0x00158d0002b7eea3
19:59:30:159 SensorNode 7 set node 0x00158d0002c7ee71
19:59:30:363 Current channel 25
19:59:30:409 CTRL got nwk update id 0
19:59:30:426 CTRL ANT_CTRL 0x03
19:59:30:538 Device protocol version: 0x0109
19:59:30:729 Current channel 25
19:59:30:777 CTRL got nwk update id 0
19:59:30:794 CTRL ANT_CTRL 0x03
19:59:33:479 scan skip host .100
19:59:33:515 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x262f0500.bin.GCF
19:59:33:515 GW firmware version: 0x262f0500
19:59:33:515 GW firmware version is up to date: 0x262f0500
19:59:33:718 Announced to internet
19:59:33:719 discovery server date: Sun, 20 Jan 2019 18:59:33 GMT
19:59:33:719 	 local time seems to be ok
19:59:33:719 discovery found version 2.04.35 for update channel stable
19:59:40:981 scan finished
19:59:43:658 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:43:658 device state timeout ignored in state 3
19:59:45:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:45:577 device state timeout ignored in state 3
19:59:46:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:46:577 device state timeout ignored in state 3
19:59:47:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:47:577 device state timeout ignored in state 3
19:59:48:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:48:577 device state timeout ignored in state 3
19:59:49:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:49:577 device state timeout (handled)
19:59:50:109 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
19:59:50:109 device state timeout ignored in state 2
19:59:50:594 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:50:594 device state timeout ignored in state 2
19:59:51:092 could not delete /var/tmp/deconz-firmware-version
19:59:51:164 Device firmware version 0x262F0500
19:59:51:209 unlocked max nodes: 200
19:59:51:492 Device protocol version: 0x0109
19:59:51:685 Current channel 25
19:59:51:732 CTRL got nwk update id 0
19:59:51:749 CTRL ANT_CTRL 0x03
19:59:51:861 Device protocol version: 0x0109
19:59:52:052 Current channel 25
19:59:52:100 CTRL got nwk update id 0
19:59:52:117 CTRL ANT_CTRL 0x03
19:59:54:269 saved node state in 1 ms
19:59:54:678 sync() in 408 ms
19:59:54:678 REST API plugin shutting down
19:59:54:778 sync() in 98 ms
```

dape82, 2019-01-21:
Does this log help or can I take any further action?

dape82, 2019-01-21:
I use a 2,5 A power supply.

[image]
[image]

What bothers me is, that it worked first. I did not change any settings. I only added sensors via the Phoscon app. I think after adding an Aquara temperature and humidity sensor it did not work any more.

dape82, 2019-01-21:
I did reset via the Phoscon App. No difference... it still disconnects...

manup, 2019-01-21:
The configuration looks valid.

Can you please also check that the serial port is configured as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

It should be, but just in case ..

Is there any other equipment connected to the RPi?

dape82, 2019-01-21:
The serial port ist configured correctly. I have a usb wureless keyboard dongle and a homematic ip module on the gpio pins: https://www.elv.de/homematic-funkmodul-fuer-raspberry-pi-bausatz.html attached. But i don't think that is the problem.

I found an old SD Card with the whole system before I upgraded to a 128 GB SD Card (I created an image from the 32GB card and copied it to the 128GB card. Then i resized the partition.)

In this old system i have no disconnects.

Are the any files I can compare or replace with the older ones?

dape82, 2019-01-21:
I copied all files in ~/.local/share/dresden-elektronik/deCONZ from the working system to the non-working system.

No effect. Still having disconnects...

dape82, 2019-01-21:
I now know which process cause the problem. It is the CuxD-Plugin (https://homematic-forum.de/forum/viewtopic.php?t=15298) of the Homematic installation (https://github.com/leonsio/YAHM).

I have no clue why that happens. Do you have any idea?

manup, 2019-01-21:
I can only guess that the plugin tries to access or probe the serial port and this scrambles the messages.

closed 2019-05-28 by stale[bot]