dresden-elektronik/deconz-rest-plugin #1159: No connection and control with 2.05.55 and firmware 262F0500

kcdr, 2019-01-20:
First it worked as with version before but now deconz does not react to commands over phoscon nor REST-api.
Versions are 2.05.55 and firmware 262F0500

In desktop-app no lines are drawn between the nodes anymore.

Console shows many such messages:
unhandled APS-DATA.confirm id: 241 status 0xE1

manup, 2019-01-20:
>unhandled APS-DATA.confirm id: 241 status 0xE1

I know this one :)

When our hardware guys jam the radio band completely Zigbee has no chance to come through.

    MAC_CHANNEL_ACCESS_FAILURE_STATUS = 0xE1,

Is there some WiFi or Bluetooth streaming ongoing?

kcdr, 2019-01-20:
Thanks for the hint.
I installed a new FritzBox yesterday. Maybe it sends on the same channel/frequency like the RaspBee.
I will try to move the raspberrypi away from it and/or change the channel.

kcdr, 2019-01-21:
Today same picture: No Connection to nodes.

Now I downgraded to 
2.05.54 and 
firmware 0x262e0500
I even switched from the new FritzBox to the old one. 
Installed a clean image of Phoscon on new sd-card.
All didn't help.

Seems last explanation is really something is in the radio band around here or the raspbee is broken.
Don't know what to do next...

kcdr, 2019-01-21:
Got it working again: Re-paired all lights, sensors and remotes.
Don't know what caused all this dilemma.

closed 2019-01-21 by kcdr

manup, 2019-01-21:
>Got it working again: Re-paired all lights, sensors and remotes.
Don't know what caused all this dilemma.

It could have been a temporary problem. Note since you did a channel change during the rf problems the gateway was likely on another channel when the devices.