dresden-elektronik/deconz-rest-plugin #1160: high Load

netzbus, 2019-01-20:
since installing this app on my Raspberry 3B+ the load is high increased
[image]

`  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                                                                                                
  304 pi        20   0  123304  38684  27592 S   2.6  4.1  11:43.97 deCONZ                                                                                                                                 
  374 root      20   0    5764   3728   2792 S   2.0  0.4   9:35.73 deCONZ-WIFI2.sh                                                                                                                        
15453 root      20   0    8244   3316   2748 R   1.0  0.3   0:02.30 top                                                                                                                                    
  471 fhem      20   0  108176  98700   9964 S   0.7 10.4  17:25.58 perl`
why is deconz wasting so much load?

manup, 2019-01-20:
Next version 2.05.57 contains some fixes for this issue.

The load should be normal again when you stop the deconz-wifi  service

    sudo systemctl stop deconz-wifi

manup, 2019-01-20:
2.05.57 is out now, it should improve the load a lot also when service would run (wip).

netzbus, 2019-01-21:
Great. Load is still normal after upgrade to 2.05.57

closed 2019-01-21 by netzbus