dresden-elektronik/deconz-rest-plugin #1161: v2.05.57 breaks rules on /config/localtime

ebaauw, 2019-01-20:
My rules on `/config/localtime` no longer fire in v2.05.57:
```
{
  "actions": [
    {
      "address": "/sensors/3/state",
      "body": {
        "flag": true
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/config/localtime",
      "operator": "in",
      "value": "T23:30:00/T07:00:00"
    }
  ],
  "created": "2018-05-12T10:51:47",
  "etag": "ad521772c5b031ebdd0c03b52bc5e580",
  "lasttriggered": "none",
  "name": "Night On",
  "owner": "0107043F97",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 0
}
```

manup, 2019-01-20:
Hmm can be only this commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/1aa3d0af9ec978de3b44321d487d199c2da7a4e3

But it should handle this fine, I'll do some tests.

manup, 2019-01-20:
Ah damn the map constructor doesn't use the default W127 when no week days are specified.

ebaauw, 2019-01-20:
What should the value be as a workaround: `"W127T23:30:00/W127T07:00:00"` or `"W127T23:30:00/T07:00:00"`?

manup, 2019-01-20:
`W127/T23:30:00/T07:00:00` should work

ebaauw, 2019-01-20:
Thanks. Will try tomorrow.

ebaauw, 2019-01-21:
Didn't try the workaround, but c245b23 fixes the issue.  Thanks.

closed 2019-01-21 by ebaauw