dresden-elektronik/deconz-rest-plugin #1164: Syslog Error: PROTO: CRC error

kugelkopf123, 2019-01-21:
Ive updated to the newest version of deconz .57 in the early morning and now i found an Error in my Syslog. Perhaps someone can tell me what it is: 

`Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 4, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 7, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 14, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 18, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 14, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 14, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
[... 113 lines cut ...]
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 7, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 14, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 18, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: ZM: Skip command with id = 23, wrong length
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
Jan 21 10:41:33 raspi4 deCONZ[341]: PROTO: CRC error
`

manup, 2019-01-21:
This refers to serial communication errors. Is deCONZ otherwise working?

adwolters, 2019-02-18:
I have the same issue. I also updated to the .57 version. I then updated to the .58 version and the problem seems to be there still. Only thing that works is a removal of the Conbee stick and re-insert it in the USB port. Then everything is working again. But after a day this needs to be repeated. 

I don't have to say how annoying this is of course. 

I tried a different USB port, but that didn't fix the issue, any other suggestions?

manup, 2019-02-18:
Can you provide more details:
* Operating system
* ConBee firmware version

Do you use a USB extension cable?

adwolters, 2019-02-18:
Sorry of course I need to provide more details. I run deconz in docker (marthoc image) on a Virtual Machine, CentOS 7. The stick has been reliable for months in this setup. 

The ConBee version is 26240500 and I just discussed the issue with a colleague of mine and he already suggested upgrading the firmware. So I am going to try that first thing and come back if the issue persists.

manup, 2019-02-18:
>The ConBee version is 26240500 and I just discussed the issue with a colleague of mine and he already suggested upgrading the firmware.

Indeed, 26240500 should be upgraded. Please make a backup first just in case.

closed 2019-06-25 by stale[bot]