dresden-elektronik/deconz-rest-plugin #1165: Xiaomi switches hourly report resends the last button event.

CultusMechanicus, 2019-01-21:
The Xiaomi battery reports on WXKG01LM switches now replay the last buttonevent, which shows up as "ghost" short releases. 
This is new in V2_05_57.
```
2019-01-21 11:11:46 DEBUG (MainThread) [pydeconz.deconzdevice] lumi.sensor_switch (2): update battery with 100
2019-01-21 11:11:46 DEBUG (MainThread) [pydeconz.deconzdevice] lumi.sensor_switch (2): update on with True
2019-01-21 11:11:46 DEBUG (MainThread) [pydeconz.deconzdevice] lumi.sensor_switch (2): update reachable with True
2019-01-21 11:11:46 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"config":{"battery":100,"on":true,"reachable":true},"e":"changed","id":"23","r":"sensors","t":"event","uniqueid":"00:15:8d:00:01:59:2f:07-01-0006"}
2019-01-21 11:11:46 DEBUG (MainThread) [pydeconz.deconzdevice] lumi.sensor_switch (2): update buttonevent with 1002
2019-01-21 11:11:46 DEBUG (MainThread) [pydeconz.deconzdevice] lumi.sensor_switch (2): update lastupdated with 2019-01-21T11:11:46
2019-01-21 11:11:46 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event deconz_event[R]: id=lumi_sensor_switch_2, event=1002>
2019-01-21 11:11:46 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"23","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2019-01-21T11:11:46"},"t":"event","uniqueid":"00:15:8d:00:01:59:2f:07-01-0006"}
```

manup, 2019-01-21:
Hmm, how is this possible, the hourly reports can't trigger button events, there is no code for this.

Haven't observed this yet, will do some tests.

CultusMechanicus, 2019-01-21:
Not sure, but all my lights are turning on anytime a button reports it's battery :) Definitely worked as expected in 05_54.

Could it be similar to the Xiaomi false motion reports that was in 05_55?

rsokolowski, 2019-01-21:
I'm observing the same issue after update to 05_57 and also using Home Assistant.

Here is log from deconz during spurious release event (id=1002).
> 10:37:06:993 no button handler for: lumi.sensor_switch ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 002
> 10:37:06:993 ZCL attribute report 0x00158D0001C30683 for cluster 0x0000, ep 0x01
> 10:37:06:993 0x00158D0001C30683 extract Xiaomi special attribute 0xFF02
> 10:37:06:993    64 on/off 1
> 10:37:06:993    01 battery 3022 (0x0BCE)
> 10:37:06:993    03 unsupported index (data type 0x21)
> 10:37:06:993    04 unsupported index (data type 0x24)
> 10:37:06:993    05 unsupported index (data type 0x21)
> 10:37:06:993    06 unsupported index (data type 0x20)
What's interesting, it doesn't mention that the button was actually released which happens if button is actually released (i.e. 12:47:45:268 button 1002 Normal release).

CultusMechanicus, 2019-01-21:
Ah, think I've worked this out. I think "lastupdated" changing now triggers a state change event, with stale data in "buttonevent".

manup, 2019-01-21:
Might be related to commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/fa5d91c26c21a9476e5eab179d2794721aca3e65

But also here is no code to reflect on button event.

CultusMechanicus, 2019-01-21:
Yep, it's probably that. 
- Issue event when `state.lastupdated` is updated to fire rules based on `state.lastupdated`.
I don't think the previous buttonevent is ever actually removed.

ebaauw, 2019-01-21:
> I don't think the previous buttonevent is ever actually removed.

There is no way to "remove" the previous button event.  `state.buttonevent` reflects the last button action that has happened and `state.lastupdated` reflects when that happened.

> Might be related to commit fa5d91c

I think so, at least the part that causes a rule to fire.  I actually think the offending code, wrongly changing `state.lastupdated` for the ZHASwitch sensor has been introduced earlier:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/fa5d91c26c21a9476e5eab179d2794721aca3e65/de_web_plugin.cpp#L8739-L8746

This doesn't check the sensor type (or rather the sensor type's resource item), causing `state.lastupdated` to be changed, even if the state isn't.  I think line 8739 should read:
```c
if (item && attrId == 0xff02)
```
We might need to check for another resource item types besides `RStateOpen` or `RStatePresence`.

CultusMechanicus, 2019-01-21:
> There is no way to "remove" the previous button event. state.buttonevent reflects the last button action that has happened and state.lastupdated reflects when that happened.

Ah, ok. That's not what's happening with the new code though. state.lastupdated reflects either the time the time of the last button push/release OR the time of the last special report. Which makes it almost impossible to work out what the last event actually was...

ebaauw, 2019-01-21:
> `state.lastupdated` reflects either the time of the last button push/release OR the time of the last special report.

Indeed.  This bug was introduced with the change to handle the special 0xff02 attribute report for the Mi devices, not sure which version that was.  While deCONZ rules wouldn’t fire before 2.05.57, automations in client apps would (incl. e.g. homebridge-hue and plugins for other home automation systems).  Also note that the last `buttonevent` would be repeated, not necessarily the 1002.  I think this explains @esseki ‘s issues.

CultusMechanicus, 2019-01-21:
Home Assistant actually wouldn't, at least not for end devices. pydeconz only uses events to trigger, which the 0xff02 didn't send until v2.05.57. 
As a quick work around on my end, I've switched to using the "initial push"/1000 event, rather than "short release"/1002, since that should basically never be the last state of the button. That does stop me using hold or multi-presses in future though.

manup, 2019-01-21:
>This doesn't check the sensor type (or rather the sensor type's resource item), causing state.lastupdated to be changed, even if the state isn't. I think line 8739 should read:

>if (item && attrId == 0xff02)
We might need to check for another resource item types besides RStateOpen or RStatePresence.

Yes this looks very much like the cause I'll put the item check inside the 0xff02 if branch to not have other side effects. Commit incoming ..

CultusMechanicus, 2019-01-31:
Looks like v2.05.58 fixed this. Happy days :)

closed 2019-01-31 by CultusMechanicus