dresden-elektronik/deconz-rest-plugin #1166: request: phoscon app add sensors mac address display

clockworkant, 2019-01-21:
Is it possible to add the mac address into the ui on the phoscon web app for sensors. I am monitoring the websockets and it would be super useful to be able to match up the mac address in the id field with the device in the phoscon app.

The two solutions I can think of is, include the mac address on the phoscon app or add the device title to the websockets output.

This is just a request that would make my life easier as I'm about to add 10 xiaomi sensors to my home!

ebaauw, 2019-01-21:
> it would be super useful to be able to match up the mac address in the id field

The `id` field in the web socket notification is the resource ID (as in `/sensors/xx`).  The mac address is the first part of the `uniqueid` attribute (as in `xx:xx:xx:xx:xx:xx:xx:xx`).

But I agree, it makes sense to be able to view the resource ID from Phoscon.  In fact, homebridge-hue exposes this to HomeKit, precisely for the purpose of matching the device to the resource and web socket notification.

manup, 2019-01-21:
When you open a sensors detail page you can see the mac address in the url bar.

[image]